pybamm-team/PyBaMM #2069: [Bug report]:  initial_soc in sim.solve() method calculating wrong SoC (seemingly because of parameters)

Mykle99, 2022-05-22:
### PyBaMM Version

22.3

### Python Version

3.9.12

### Describe the bug

initial_soc does not set the correct SoC value when: 
- Not specifying any params
- Using different base param sets (such as Mohtat2020 and Chen2020), yields different SoC for the same initial_soc value

### Steps to Reproduce

Can be reproduced by testing different params```
experiment = pybamm.Experiment([
        ("Rest for 1 hour")]   
)

chen2020_params = pybamm.parameter_sets.Chen2020
params = pybamm.ParameterValues(chemistry = chen2020_params)

lower_SoC = 0.2
spme = pybamm.lithium_ion.SPMe()
sim = pybamm.Simulation(spme, parameter_values = params, experiment = experiment)
sol = sim.solve(initial_soc = lower_SoC)

ne_soc = sol["Negative electrode SOC"].data
print("Expected SoC: ", lower_SoC, "Actual NE SoC: ", ne_soc)
```

### Relevant log output

```shell
initial_soc = 0.2

# Mistake seemingly has nothing to do with base battery model
# SPMe with submodel options:    0.2 -> 0.18010387
# SPMe without submodel options: 0.2 -> 0.18010387
# DFN with submodel options:     0.2 -> 0.18010387
# DFN without submodel options:  0.2 -> 0.18010387 

# Mistake seemingly has nothing to do with which submodels are used
# SPMe submodel options (liplt, SEI, particle mechanics, thermal, LAM): 0.2 -> 0.18010387 
# SPMe without specifying solver: 0.2 -> 0.18010387
# SPMe without specifying var_pts: 0.2 -> 0.18010387

# Parameters are the issue:
# SPMe no custom options, no specified params: 0.2 -> 0.33643954 

# initial_soc = 0.2 is only correct when applying Chen2020, which might be random,
# because Mohtat2020 gives wrong SoC. 
# SPMe no custom options, using Chen2020 base param set: 0.2 -> 0.20320027 
# SPme no custom options, using Mohtat2020 base param set: 0.2 -> 0.16788433
```

tinosulzer, 2022-05-22:
Negative(/Positive) electrode SOC is not the same as cell SOC. See below, where `z` is cell SOC, `x` is negative electrode SOC, `y` is positive electrode SOC. There is no variable for cell SOC, because calculating it is kind of hard / unstable
[image]
/

closed 2022-05-22 by tinosulzer

Mykle99, 2022-05-22:
Thank you for clearing up my misconception! With your explanation In mind, the changing NE SoC `x` is counteracted by a similar change in PE SoC `y`, such that `z` satisfies what is set by `initial_soc`?