pybamm-team/PyBaMM #2072: [Bug]: on installation, "No Module named 'CMakeBuild'"

khurlbutt, 2022-05-23:
### PyBaMM Version

latest

### Python Version

3.8.9

### Describe the bug

I am trying to install PyBaMM on Mac OS. Running

-> brew install sundials && pip3 install pybamm

gives the error

"ModuleNotFoundError: No module named 'CMakeBuild'".

This is after succesfful install / update to sundials and during the attempt to install pybamm itself. Any advice or guidance on how to get pybamm installed would be much appreciated.

### Steps to Reproduce

In terminal on Mac OSx 12.2 running Python 3.8 issue "brew install sundials && pip3 install pybamm".

### Relevant log output

```shell
ModuleNotFoundError: No module named 'CMakeBuild'
```

martinjrobins, 2022-05-23:
Hi @khurlbutt, can you please post your full log output?

khurlbutt, 2022-05-24:
@martinjrobins it begins below (as a reminder, I issued "brew install sundials && pip3 install pybamm")

###BEGIN OUTPUT###
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
Updated 1 formula.

Warning: sundials 6.2.0 is already installed and up-to-date.
To reinstall 6.2.0, run:
  brew reinstall sundials
Defaulting to user installation because normal site-packages is not writeable
Collecting pybamm
  Using cached pybamm-22.4.tar.gz (908 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy>=1.16 in ./Library/Python/3.8/lib/python/site-packages (from pybamm) (1.22.2)
Requirement already satisfied: scipy>=1.3 in ./Library/Python/3.8/lib/python/site-packages (from pybamm) (1.8.0)
Collecting pandas>=0.24
  Using cached pandas-1.4.2-cp38-cp38-macosx_11_0_arm64.whl (9.9 MB)
Collecting anytree>=2.4.3
  Using cached anytree-2.8.0-py2.py3-none-any.whl (41 kB)
Collecting autograd>=1.2
  Using cached autograd-1.4-py3-none-any.whl (48 kB)
Collecting scikit-fem>=0.2.0
  Using cached scikit_fem-6.0.0-py3-none-any.whl (144 kB)
Collecting pybamm
  Using cached pybamm-22.3.tar.gz (907 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-22.2.tar.gz (907 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-22.1.tar.gz (906 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.12.tar.gz (907 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.11.tar.gz (901 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.10.tar.gz (897 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.9.tar.gz (894 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.8.post2.tar.gz (890 kB)
  Preparing metadata (setup.py) ... done
  Using cached pybamm-21.8.post1.tar.gz (888 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/private/var/folders/w8/w8ymv3_s30z1s4rccl6ypmgm0000gn/T/pip-install-pqy0bl37/pybamm_0fcd5ef3354e4db68c3cfe53f753d1ab/setup.py", line 18, in <module>
          import CMakeBuild
      ModuleNotFoundError: No module named 'CMakeBuild'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
###END OUTPUT###

martinjrobins, 2022-05-24:
It looks like you have one of the newer apple M1 CPUs? If so we don't support these yet via installation from PyPI. I would recommend that you try installing from source, using the instructions here: https://pybamm.readthedocs.io/en/latest/install/install-from-source.html

martinjrobins, 2022-05-26:
its not finding casadi in your python environment. Can you try `pip install casadi` before you run this command

tinosulzer, 2022-09-27:
Has this been solved?

jaskiratsingh2000, 2023-02-20:
Is this still open?

tinosulzer, 2023-02-20:
We'll close it since we haven't heard back in a while

closed 2023-02-20 by tinosulzer