pybamm-team/PyBaMM #2097: [Bug]: `IDAKLUSolver` not accessible from `PyBaMM`'s directory in `Windows` and `WSL` (when installed using `pip`)

Saransh-cpp, 2022-06-09:
### PyBaMM Version

develop

### Python Version

3.9.0

### Describe the bug

The `IDAKLUSolver` works in `Windows` and `WSL` but with a catch - I cannot use the solver if I run a script in `PyBaMM`'s root directory. I can use it everywhere else, even in the subdirectories within the `PyBaMM` root directory. Due to this bug, I also cannot run the tests intended for `IDAKLUSolver` as they throw an error saying -
```
skipped 'idaklu solver is not installed'
```
The error when using the solver from `PyBaMM`'s directory -
```
>>> import pybamm
>>> pybamm.__version__
'22.5'
>>> pybamm.IDAKLUSolver()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/c/Users/Saransh/Saransh_softwares/OpenSource/Python/PyBaMM/pybamm/solvers/idaklu_solver.py", line 58, in __init__
    raise ImportError("KLU is not installed")
ImportError: KLU is not installed
```
Now, there are 2 cases -
1. Installing from pip
The error is the same for both `Windows` and `WSL`.
2. Installing from source on `WSL`
This works! I can use the `IDAKLUSolver` from any directory, and I can also run the tests.

### Steps to Reproduce

The steps are exactly the same as in the case of using `pip` for installation in `WSL`.

```
(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>python -m pip freeze | findstr pybamm

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>python -m pip install pybamm --no-cache

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>python -m pip freeze | findstr pybamm
pybamm==22.5

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>python
Python 3.9.0 (tags/v3.9.0:9cf6752, Oct  5 2020, 15:34:40) [MSC v.1927 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pybamm
>>> pybamm.IDAKLUSolver()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM\pybamm\solvers\idaklu_solver.py", line 58, in __init__
    raise ImportError("KLU is not installed")
ImportError: KLU is not installed
>>> exit()

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>cd ..

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python>python
Python 3.9.0 (tags/v3.9.0:9cf6752, Oct  5 2020, 15:34:40) [MSC v.1927 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pybamm
>>> pybamm.IDAKLUSolver()
<pybamm.solvers.idaklu_solver.IDAKLUSolver object at 0x0000024F6AC78BB0>
>>> exit()

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python>cd PyBaMM

(env) C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM>python -m unittest discover tests\unit\test_solvers -v
# skipping to the IDAKLUSolver tests ->

...
test_sensitivities (test_base_solver.TestBaseSolver) ... skipped 'idaklu solver is not installed'
...
test_dae_solver_algebraic_model (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_failures (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_ida_roberts_klu (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_ida_roberts_klu_sensitivities (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_input_params (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_model_events (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
test_set_atol (test_idaklu_solver.TestIDAKLUSolver) ... skipped 'idaklu solver is not installed'
...
```

martinjrobins, 2022-06-11:
I think what is happening is that when you have pip installed pybamm, you have 2 versions of pybamm. The pip installed version living in your `site-packages`, and the source directory living at `C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM`. When your current working directory is `C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM`, then when you run a test or script using pybamm, it imports the pybamm in your current directory, i.e. `C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM\pybamm`, not the one in `site-packages`. Is the KLU solver installed in `C:\Users\Saransh\Saransh_softwares\OpenSource\Python\PyBaMM\pybamm`? If not then this might be the source of your problem.

Saransh-cpp, 2022-06-28:
Oh yes, this makes sense. I went through the installation again, and the python interpreter does import pybamm from the current working directory. Thanks! I can still make it work in WSL!

closed 2022-06-28 by Saransh-cpp