pybamm-team/PyBaMM #2101: [Bug]: Running an experiment simulation changes future simulations outputs

cannotcomplain, 2022-06-09:
### PyBaMM Version

22.3

### Python Version

3.9.12

### Describe the bug

Using a DFN model, if wo identical constant current discharge simulations are solved (sim1 and sim3), each of which produces the same output.

But, if the same (identical) simulations are solved but a separate simulation is run in-between (sim2) using an experiment definition, then the two solutions of sim1 and sim3 are different.

The solutions of sim1 and sim3 are equal  if any of the following are true:
a) the simulation is pre-defined (e.g. sim = pybamm.Simulation(...)) and then before and after the experiment the solutions are produced with sim1 = sim.solve
b) The in-between simulation (sim2) is not defined with an experiment (e.g. a current function or defined C_rate)

### Steps to Reproduce

import pybamm
import numpy as np
pybamm.set_logging_level(100)

model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues('Ramadass2004')

t = np.arange(0, 7200, 0.5)
solver = pybamm.CasadiSolver(atol=1e-6, rtol=1e-6, root_method='lm', root_tol=1e-6, dt_max=5)
#%%
sol1 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)
sol3 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)

print('These are the same')
print(f"sol1 final voltage = {sol1['Terminal voltage [V]'].data[-1]:.5f} V")
print(f"sol3 final voltage = {sol3['Terminal voltage [V]'].data[-1]:.5f} V")

sol1 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)

sol2 = pybamm.Simulation(model, parameter_values=params, experiment=pybamm.Experiment([
    'Discharge at 1 C until 2.8 V'
])).solve(solver=solver)

sol3 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)

print('One of these things is not like the other one')
print(f"sol1 final voltage = {sol1['Terminal voltage [V]'].data[-1]:.5f} V")
print(f"sol3 final voltage = {sol3['Terminal voltage [V]'].data[-1]:.5f} V")

### Relevant log output

```shell
These are the same
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.80000 V

One of these things is not like the other one
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.77431 V
```

brosaplanella, 2022-06-10:
What's the termination event in `sol3` in the two scenarios?

cannotcomplain, 2022-06-10:
It should be the lower cut-off voltage 2.8 V

brosaplanella, 2022-06-10:
It could be that defining the simulation 2 removes an event. Can you print `sol3.model.events` in both cases?

cannotcomplain, 2022-06-10:
I get an `AttributeError` for `sol3.model` but, if I run 
```
sim3 = pybamm.Simulation(model, parameter_values=params)
sol3 = sim3.solve(t, solver=solver, initial_soc=1)
[print(x.name) for x in sim3.model.events];
```

I get the following lists...
First example:
```
Zero negative electrode porosity cut-off
Max negative electrode porosity cut-off
Zero positive electrode porosity cut-off
Max positive electrode porosity cut-off
Zero electrolyte concentration cut-off
Minimum voltage
Maximum voltage
Minimum voltage switch
Maximum voltage switch
These are the same
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.80000 V
```
Second example:
```
Zero negative electrode porosity cut-off
Max negative electrode porosity cut-off
Zero positive electrode porosity cut-off
Max positive electrode porosity cut-off
Zero electrolyte concentration cut-off
Minimum voltage
Maximum voltage
Minimum voltage switch
Maximum voltage switch
One of these things is not like the other one
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.77431 V
```

cannotcomplain, 2022-06-10:
I did some more digging into the events, and I think this is more of what you were asking...

I compared the `id` of the events for the two simulations and two of them change after the experiment run `Minimum voltage` and `Maximum voltage`

Printing the expressions for each event, before and after the middle experiment, highlights the problem.

The two expressions for `Minimum voltage` are shown below, the difference is after the experiment simulation the expression has a $ + 1.0 $ at the very end that wasn't there before. I've confirmed that if you continue to run experiment simulations, this value increments up (e.g. $+ 2.0$, $ + 3.0$) for each experiment performed. 

The opposite happens for the 'Maximum voltage' events, which decreases by $1.0$ each time an experiment is run.

Here is the code snippet...
```
import pybamm
import numpy as np
pybamm.set_logging_level(100)

model = pybamm.lithium_ion.DFN()
params = pybamm.ParameterValues('Ramadass2004')

t = np.arange(0, 7200, 0.5)
solver = pybamm.CasadiSolver(atol=1e-6, rtol=1e-6, root_method='lm', root_tol=1e-6, dt_max=5)
#%%
# Example 1 - No intermediate experiment
print('\n--Example 1 - without intermediate simulation')
sim1 = pybamm.Simulation(model, parameter_values=params)
sol1 = sim1.solve(t, solver=solver, initial_soc=1)

sim3 = pybamm.Simulation(model, parameter_values=params)
sol3 = sim3.solve(t, solver=solver, initial_soc=1)
# for x in sim3.model.events: print(x)
# print(sol3.model.events)
print('These are the same')
print(f"sol1 final voltage = {sol1['Terminal voltage [V]'].data[-1]:.5f} V")
print(f"sol3 final voltage = {sol3['Terminal voltage [V]'].data[-1]:.5f} V")

# Example 2 - With intermediate simulation
print('\n\n --Example 2 - with intermediate simulation')
sol1 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)
print('')
print(model.events[5].name)
print(model.events[5].expression)
sol2 = pybamm.Simulation(model, parameter_values=params, experiment=pybamm.Experiment([
    'Discharge at 1 C until 2.8 V'
])).solve(solver=solver)

# sol3 = pybamm.Simulation(model, parameter_values=params).solve(t, solver=solver, initial_soc=1)
sim3 = pybamm.Simulation(model, parameter_values=params)
sol3 = sim3.solve(t, solver=solver, initial_soc=1)
# for x in sim3.model.events: print(x)
print('')
print(model.events[5].name)
print(model.events[5].expression)

print('One of these things is not like the other one')
print(f"sol1 final voltage = {sol1['Terminal voltage [V]'].data[-1]:.5f} V")
print(f"sol3 final voltage = {sol3['Terminal voltage [V]'].data[-1]:.5f} V")
```

Here is the output snippet....
```
--Example 1 - without intermediate simulation
These are the same
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.80000 V

 --Example 2 - with intermediate simulation

Minimum voltage
boundary value(boundary value(Positive electrode potential)) - ((Lower voltage cut-off [V] - (Positive electrode OCP [V] + 1e-06 * 1.0 / yz-average(x-average(r-average(Initial concentration in positive electrode [mol.m-3] / Maximum concentration in positive electrode [mol.m-3]))) + 1e-06 * 1.0 / (yz-average(x-average(r-average(Initial concentration in positive electrode [mol.m-3] / Maximum concentration in positive electrode [mol.m-3]))) - 1.0) - (Negative electrode OCP [V] + 1e-06 * 1.0 / yz-average(x-average(r-average(Initial concentration in negative electrode [mol.m-3] / Maximum concentration in negative electrode [mol.m-3]))) + 1e-06 * 1.0 / (yz-average(x-average(r-average(Initial concentration in negative electrode [mol.m-3] / Maximum concentration in negative electrode [mol.m-3]))) - 1.0)))) / (8.617333262281981e-05 * Reference temperature [K]))

Minimum voltage
boundary value(boundary value(Positive electrode potential)) - ((Lower voltage cut-off [V] - (Positive electrode OCP [V] + 1e-06 * 1.0 / yz-average(x-average(r-average(Initial concentration in positive electrode [mol.m-3] / Maximum concentration in positive electrode [mol.m-3]))) + 1e-06 * 1.0 / (yz-average(x-average(r-average(Initial concentration in positive electrode [mol.m-3] / Maximum concentration in positive electrode [mol.m-3]))) - 1.0) - (Negative electrode OCP [V] + 1e-06 * 1.0 / yz-average(x-average(r-average(Initial concentration in negative electrode [mol.m-3] / Maximum concentration in negative electrode [mol.m-3]))) + 1e-06 * 1.0 / (yz-average(x-average(r-average(Initial concentration in negative electrode [mol.m-3] / Maximum concentration in negative electrode [mol.m-3]))) - 1.0)))) / (8.617333262281981e-05 * Reference temperature [K])) + 1.0
One of these things is not like the other one
sol1 final voltage = 2.80000 V
sol3 final voltage = 2.77431 V
```

cannotcomplain, 2022-06-13:
It looks like this behavior was explicitly defined in line 407 of `pybamm/simulation.py`
```python
                # Keep the min and max voltages as safeguards but add some tolerances
                # so that they are not triggered before the voltage limits in the
                # experiment
                for event in new_model.events:
                    if event.name == "Minimum voltage":
                        event._expression += 1
                    elif event.name == "Maximum voltage":
                        event._expression -= 1
```

Is there a simple way to reset the events after a simulation?

tinosulzer, 2022-06-17:
There are a few workarounds possible (the easiest being to just define the model again), but the real fix would be to replace
```python
for event in new_model.events:
                    if event.name == "Minimum voltage":
                        event._expression += 1
                    elif event.name == "Maximum voltage":
                        event._expression -= 1
```
with something like
```python
for i, event in enumerate(new_model.events):
                    if event.name == "Minimum voltage":
                        new_model.events[i] = pybamm.Event(event.name, event._expression + 1, event._event_type)
                    elif event.name == "Maximum voltage":
                        new_model.events[i] = pybamm.Event(event.name, event._expression - 1, event._event_type)
```
This is because the bug is due to a shallow copy instead of deep copy in `BaseModel.new_copy()`. I haven't tried this but I think it should work, can you test it and open a PR if it does?

tinosulzer, 2022-07-01:
#2139 should fix this

closed 2022-07-01 by tinosulzer

cannotcomplain, 2022-07-01:
That's great! Thanks!

PR #2139 (2022-07-01): Issue 2124 bug
# Description

Some improvements to diagnosing solver errors:
- Adds event for the particle cracking submodel when the crack is larger than the particle radius
- Adds logger error when a state vector exceeds 1e5

Fixes #2124 
Fixes #2138 
Fixes #2101 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionalit [... cut]