pybamm-team/PyBaMM #2103: Battery aging experiment within different SOC ranges.

zhunxinren, 2022-06-15:
### Description

Cycle the battery within different SOC ranges with the accelerated aging tests. For example, cycle the battery within 20%-80% for the aging test.

### Motivation

It is widely known that the battery ages at different rates under different SOC ranges. This feature will help to further investigate the aging mechanisms when the cycling SOC is different.

rtimms, 2022-06-19:
See [this slack post](https://pybamm.slack.com/archives/CJ19X0JCF/p1650896004915399)

Sourabh-Jadhav-2, 2022-08-22:
I am not able to open this post

rtimms, 2023-05-15:
Lots of discussions related to this, e.g. [here](https://github.com/pybamm-team/PyBaMM/discussions/2936). Closing.

closed 2023-05-15 by rtimms