pybamm-team/PyBaMM #2111: Adding customized function for tortuosity

amirsh213, 2022-06-17:
### Description

I was wondering if it is possible to enable adding customized non-Bruggeman functions to define tortuosity or MacMullin number as a function of porosity?

Thanks.

rtimms, 2022-06-19:
You could create a new submodel in [this folder](https://github.com/pybamm-team/PyBaMM/tree/develop/pybamm/models/submodels/transport_efficiency). The simplest thing would be to copy the Bruggeman submodel and edit it to whatever function you want.

rtimms, 2022-06-23:
I'm currently working on adding extra documentation around how to use submodels, but there is an example of swapping out submodels [here](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/models/using-submodels.ipynb). If you wanted to integrate your new submodel into the code you could make it so that you pass an option to the model to set the tortuosity. Take a look at the [base battery model](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/base_battery_model.py) to see how options are handled.

For `FunctionParameters` you could look at how [particle radius](https://github.com/pybamm-team/PyBaMM/blob/0fd8cf360725ffdc5cd177de60ec78ad79833ac6/pybamm/parameters/geometric_parameters.py#L123) is defined, and see [this example](https://github.com/pybamm-team/PyBaMM/blob/develop/examples/scripts/compare_lithium_ion_particle_distribution.py) for how to provide a function. Yes, you can define your new parameters in `GeometricParameters`.

See [this discussion](https://github.com/pybamm-team/PyBaMM/discussions/1712) around how the properties are defined (we actually define a "transport efficiency", not a tortuosity.

TomTranter, 2023-01-31:
Hi @amirsh213 , how's it going? We are running a hackathon on 29th of March in Oxford and this issue is potentially one we could work on together. You'd be welcome to join us if you can make it. Tickets are free but limited availability https://www.eventbrite.co.uk/e/pybamm-hackathon-tickets-517334430207?keep_tld=1
If you have already made some progress on this issue then let us know and we will exclude it from the options to work on during the day. Thanks

PR #2841 (2023-10-11): WIP: transport efficiency models
# Description

Adding new methods for calculating the transport efficiency beyond Bruggeman coefficient. This includes:

- log square root
- log
- linear
- half volume fraction
- Bruggeman
- Tortuosity factor

These were taken from 10.1016/j.ces.2007.03.041 and http://dx.doi.org/10.1016/j.coche.2016.02.006

Fixes #2111 

#

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to th [... cut]
PR #3437 (2024-05-13): Issue 2111 custom transport efficiency
# Description

Adding new methods for calculating the transport efficiency beyond Bruggeman coefficient. This includes using the following relations for tortuosity factor ($\tau$) for use in transport efficiency $B=\epsilon/\tau$:

[image]

- Tortuosity factor: allows direct specification of a tortuosity factor
These were taken from 10.1016/j.ces.2007.03.041 and http://dx.doi.org/10.1016/j.coche.2016.02.006

Fixes #2111 

#

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CH [... cut]