pybamm-team/PyBaMM #2124: Error while simulating cycling

amirsh213, 2022-06-27:
### PyBaMM Version

22.5

### Python Version

3.8.10

### Describe the bug

I'm struggling with fast-charge (2C) aging (SEI+irreversible plating) simulation for long cycling like 50 cycles or above. I tried different things like:
- Different models: DFN, SPMe
- Different ambient temperatures: 0-25 C
- Different i0s for SEI and plating
- Different solver modes and dt_max: Only Casadi "safe" works

The problem has two parts: first, I could run it in COMSOL with more aggressive parameters e.g. higher anode tortuosity, lower temperatures, higher charge C-rates, etc.; however, in Pybamm, I am getting the error concentration reached zero with much smaller tortuosity. 
Second, when I use smaller anode tortuosity it works without the irreversible Li plating submodel only if I include CV discharge at 2.5V at the end, otherwise, it raises the error "Experiment is infeasible" in the middle of cycling and stops at cycle 26 (reversible plating also works fine in this case). So, the problem comes from the irreversible Li plating model. I use the O'Kane model and parameters but it seems that Pybamm has some issues with it although LLI is not big at the end. The script and error for 25C are below:

### Steps to Reproduce

```
# Solving for 2C fast charge

parameter_values["Initial concentration in negative electrode [mol.m-3]"] = 300
parameter_values["Initial concentration in positive electrode [mol.m-3]"] = 49000

model = pybamm.lithium_ion.DFN({"thermal": "x-full", "SEI": "solvent-diffusion limited","lithium plating":"irreversible"})

solver= pybamm.CasadiSolver("safe")

experiment = pybamm.Experiment(
    [
        
            "Charge at 2 C until 4.2 V",
            "Hold at 4.2V until C/20",
            "Rest for 15 minutes",
            "Discharge at 0.5 C until 2.5 V",
            "Hold at 2.5 V until C/20",
            "Rest for 15 minutes"
        
    ]
    * 50)

var_pts = {
    "x_n": 40,  # negative electrode
    "x_s": 5,  # separator 
    "x_p": 10,  # positive electrode
    "r_n": 20,  # negative particle
    "r_p": 10,  # positive particle
}

sim = pybamm.Simulation(model, var_pts=var_pts, experiment=experiment, parameter_values=parameter_values, solver=solver)

sim.solve(calc_esoh=False)
sim.plot();

##################
At t = 0.00934147 and h = 1.58982e-018, the corrector convergence failed repeatedly or with |h| = hmin.
At t = 0.00949702, , mxstep steps taken before reaching tout.
At t = 0.00854179 and h = 6.02689e-026, the corrector convergence failed repeatedly or with |h| = hmin.
The linesearch algorithm failed with too small a step.
2022-06-26 22:07:10.050 - [WARNING] callbacks.on_experiment_infeasible(237): 

	Experiment is infeasible: 'final time' was triggered during 'Discharge at 0.5 C until 2.5 V'. The returned solution only contains the first 153 cycles. Try reducing the current, shortening the time interval, or reducing the period.
```

Luo630, 2022-06-28:
Hi, I have this problem too, have you solved it?

amirsh213, 2022-06-28:
I just assign the chemistry before the posted script and parameterize the model; that's it all. I noticed that if I decrease the Lithium plating kinetic rate constant [m.s-1] by an order of magnitude (1e-11) it could run 50 cycles (0.8% LLI). With 1e-10 I get 3.8% LLI at cycle 26 but the LLI that I got from parameterization in COMSOL is much bigger than this. 

```
chemistry=pybamm.parameter_sets.Chen2020 # starting from an existing chemistry
# Updating the chemistry
chemistry["cell"]= "cell_pouch1"
chemistry["negative electrode"]= "graphite_pouch1"
chemistry["separator"]= "separator_pouch1"
chemistry["positive electrode"]= "nmc_pouch1"
chemistry["electrolyte"]= "electrolyte_pouch1"
chemistry["sei"]= "sei_pouch1"
chemistry["experiment"]= "experiment_pouch1"
chemistry["lithium plating"]= "plating_pouch1"

parameter_values = pybamm.ParameterValues(chemistry=chemistry)
```

Luo630, 2022-06-29:
When I tried to combine SEI film growth and particle cracking, I used Chen's parameters to simulate the experiment and found that the experiment was terminated at the 78th cycle because the corrector convergence failed repeatedly or with |h| = hmin, but I wanted to explore the battery decay of capacity after 1000 cycles，The script and error  are below:                              
  import pybamm

pybamm.set_logging_level("NOTICE")
option={"particle": "Fickian diffusion",
        "SEI film resistance":"distributed",
        "SEI":"solvent-diffusion limited",
        "particle mechanics": "swelling and cracking"
        }
model_crack=pybamm.lithium_ion.DFN(options=option)

chen_params = pybamm.ParameterValues("Chen2020")
ai_params= pybamm.ParameterValues("Ai2020")
for key, value in ai_params.items():
    if key not in chen_params.keys():
        chen_params.update({key: value}, check_already_exists=False)
chen_params.update({'Negative electrode cracking rate':3.9E-20,'Positive electrode cracking rate':3.9E-20})

experiment1=pybamm.Experiment([
        "Hold at 4.2V until 50mA",
        "Rest for 4 hours",
        ])
solver=pybamm.CasadiSolver(mode="fast with events",atol=1e-5,rtol=1e-5)
var_pts = {"x_n": 40, "x_s":40, "x_p": 40, "r_n": 40, "r_p":40}
sim_crack1=pybamm.Simulation(model=model_crack,parameter_values=chen_params,experiment=experiment1,solver=solver,var_pts=var_pts)
sim_crack1.solve()
solution1=sim_crack1.solution

experiment2=pybamm.Experiment([
        "Discharge at 1C until 2.5V",
        "Charge at 0.3C until 4.2V",
        "Hold at 4.2V until 50mA",
        ]*100)
sim_crack2=pybamm.Simulation(model=model_crack,parameter_values=chen_params,experiment=experiment2,solver=solver,var_pts=var_pts)
sim_crack2.solve(starting_solution=solution1)

solution2=sim_crack2.solution
#discharge_capacity=list(solution2.summary_variables["Capacity [A.h]"].data)
#print(discharge_capacity)
solution2.plot()
————————————————————————————————————————————————————————psetup failed: .../casadi/interfaces/sundials/idas_interface.cpp:849: Calculating Jacobian failed
At t = 0.178162 and h = 1.01303e-034, the corrector convergence failed repeatedly or with |h| = hmin.
2022-06-29 09:29:29.921 - [WARNING] callbacks.on_experiment_infeasible(237): 

	Experiment is infeasible: 'event: Current cut-off (negative) [A] [experiment]' was triggered during 'Discharge at 1C until 2.5V'. The returned solution only contains the first 236 cycles. Try reducing the current, shortening the time interval, or reducing the period.

Luo630, 2022-06-30:
There may be other reasons I think, because I updated the parameter set of Chen2020 with the parameter set of Ai2020.

brosaplanella, 2022-06-30:
Yes, but by patching two different sets the results might not be physical anymore and that's why it stops after a few cycles. Have you tried plotting the solution for the first few cycles?

Luo630, 2022-06-30:
Automatically generated, I am not very clear which eigenvalue to draw to find the problem.

brosaplanella, 2022-06-30:
I mean plotting the various internal states (concentrations, potentials...) to understand what's going on for the physical point of view.

Luo630, 2022-06-30:
I will try it, thanks

amirsh213, 2022-06-30:
Besides aging in cycling, I'm wondering why concentration goes to zero for 2C charge at room temp? Just to make sure, you defined your effective properties as e.g. `D_eff=D_int*eps**b_e` meaning the MacMullin number is `Nm=1/eps**b_e`?

tinosulzer, 2022-06-30:
It depends on what parameters you use. For example, the Chen2020 cell is an energy cell and is only rated to go up to 1.5C https://github.com/tinosulzer/CMU_energy_storage_HW5/blob/main/parameters/specsheet.pdf

amirsh213, 2022-07-01:
That's right but I could run it in COMSOL without any problem and even with more severe conditions. Can you please confirm the second part about effective properties?

tinosulzer, 2022-07-01:
Yes, that is correct

PR #2139 (2022-07-01): Issue 2124 bug
# Description

Some improvements to diagnosing solver errors:
- Adds event for the particle cracking submodel when the crack is larger than the particle radius
- Adds logger error when a state vector exceeds 1e5

Fixes #2124 
Fixes #2138 
Fixes #2101 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionalit [... cut]
closed 2022-07-01 by tinosulzer