pybamm-team/PyBaMM #2136: Charging current controlled depends upon the battery temperature

Sourabh-Jadhav-2, 2022-06-30:
### Description

Please add a feature for the charging, advanced charging method, in which the charging current is changed depending upon the battery temperature

### Motivation

[Uploading CT_CV_algorithm.pdf…]()
Please see the attached research paper

tinosulzer, 2022-06-30:
The link you shared doesn't work

Sourabh-Jadhav-2, 2022-06-30:
> The link you shared doesn't work

https://ieeexplore.ieee.org/abstract/document/9070505

tinosulzer, 2022-06-30:
This looks maybe doable, but I doubt it will be a priority for any of the existing developers so you would need to do it yourself if you want it done any time soon

Sourabh-Jadhav-2, 2022-06-30:
> This looks maybe doable, but I doubt it will be a priority for any of the existing developers so you would need to do it yourself if you want it done any time soon

I am doing it, but I just give a suggestion to you as a battery engineer.

tinosulzer, 2023-05-08:
Closing due to lack of updates, feel free to reopen if still active

closed 2023-05-08 by tinosulzer