pybamm-team/PyBaMM #2138: Raise error or warning when one of the state variables gets too large

tinosulzer, 2022-06-30:
We should identify if a state variable gets too large (e.g. 1e10) as this indicates that either the model has been badly non-dimensionalized, or the model is wrong, or the parameters are wrong. This would give a clearer error message for cases like #2124 
Should this be an error or warning? Maybe a logger error (doesn't cause the code to fail)?

brosaplanella, 2022-07-01:
Warning is probably best

tinosulzer, 2022-07-01:
How about an error, but with a setting for the max allowed value that the user can change? I think it is quite serious if `y` gets that large, and should be treated as such, but with some flexibility to allow larger values by changing the setting.

brosaplanella, 2022-07-01:
That sounds reasonable. Are you thinking of a setting like defining the smoothness of `max` and similar functions? That would be useful I think.

tinosulzer, 2022-07-01:
Yeah, see #2139

PR #2139 (2022-07-01): Issue 2124 bug
# Description

Some improvements to diagnosing solver errors:
- Adds event for the particle cracking submodel when the crack is larger than the particle radius
- Adds logger error when a state vector exceeds 1e5

Fixes #2124 
Fixes #2138 
Fixes #2101 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionalit [... cut]
closed 2022-07-01 by tinosulzer