pybamm-team/PyBaMM #2153: Discussion model benchmarks

brosaplanella, 2022-07-05:
This issue is to discuss ideas around standard model benchmarks. Here is my suggestion, as a starting point for the discussion. The main idea is to group benchmarks for a specific model, and by a specific model I mean with all the options (e.g. thermal, degradation...) fixed. Then, for each model we test various:
* Parameter sets
* Mesh sizes
* Solvers & settings
* CC discharge and various experiments

I think for the moment we can just change one setting at the time (the default could be the first setting of each group). We can later think how to sweep two or more variables (e.g. parameter sets & mesh sizes).

In terms of implementation, we can define a base class to which we can pass the model we want to study, and all benchmarks can be inherited from there. If we keep notation consistent, it should be fairly easy to write a script to process the JSON files.

tinosulzer, 2022-11-09:
What's the plan with this?

brosaplanella, 2022-11-10:
I guess it is on hold (I opened this issue more as a discussion rather than a specific task). Saying that, I realise we should probably convert it into a discussion and follow it up there.