pybamm-team/PyBaMM #2162: [Bug]: Error when solving simulation for SPM with modified Prada2013 parameter set

efg567, 2022-07-07:
### PyBaMM Version

22.4

### Python Version

3.9.13

### Describe the bug

I need to simulate the life expectancy of cells for a project. In doing so, I run into difficulties with the LFP cells. I am using the Prada2013 parameter set here, as this is the only set that implements an LFP cell.
To age the cell during the experiment, I add the same SEI used in Chen2020.
I control the aging of the cell by adjusting the parameter "SEI kinetic rate constant [m.s-1]" and setting the parameter "SEI open-circuit potential [V]" to zero, as it affects the aging too much (at least this was the case for the NMC cells where I used the Chen2020 parameter set). Unfortunately, Pybamm keeps crashing with the modified parameter set.
I also tried to upgrade to the latest Pybamm (version 22.6) and run the code. In this case, Pybamm keeps crashing even when I just try to solve the simulation with an unchanged parameter set of Prada2013.

### Steps to Reproduce

```
import pybamm
import numpy as np

def get_Wh(solution: pybamm.Solution) -> tuple:
    """Returns the Wh based on pybamm Solution. Wh are calculated from current and Voltage.

    Args:
        solution (pybamm.Solution): Pybamm solution.

    Returns:
        tuple: Wh discharged and Wh charged
    """
    t_array = solution["Time [h]"].entries
    I_array = solution["Current [A]"].entries
    U_array = solution["Battery voltage [V]"].entries

    d_t = np.diff(t_array)
    current = np.convolve(I_array, np.ones(2) / 2, mode="valid") * (-1)
    current_out = np.where(current < 0, current, 0)
    voltage = np.convolve(U_array, np.ones(2) / 2, mode="valid")

    Wh_out = abs(np.sum(voltage * current_out * d_t))

    return Wh_out

pybamm.set_logging_level("NOTICE")

ref_model = pybamm.lithium_ion.SPM()

ref_param_set = pybamm.parameter_sets.Prada2013
ref_parameter_values = pybamm.ParameterValues(chemistry=ref_param_set)

ref_experiment = pybamm.Experiment(
    [
        (
            "Discharge at 0.1 C until 2.0 V",
            "Rest for 10 minutes"
        )
    ] +
    [
        (
            "Charge at 0.1 C until 3.6 V",
            "Hold at 3.6 V until 50mA",
            "Discharge at 0.1 C until 2.0 V"
        )
    ]
)

ref_sim = pybamm.Simulation(ref_model, ref_experiment, parameter_values=ref_parameter_values)
ref_sim.solve(solver=pybamm.CasadiSolver(mode="safe"))
ref_solution = ref_sim.solution.cycles[-1]

cycle_model = pybamm.lithium_ion.SPM(
    options={
        "SEI": "ec reaction limited",
        "SEI film resistance": "distributed",
        "SEI porosity change": "false",
    }
)

cycle_param_set = pybamm.parameter_sets.Prada2013
cycle_param_set["sei"] = "example"
cycle_parameter_values = pybamm.ParameterValues(chemistry=cycle_param_set)
cycle_parameter_values.update({"SEI open-circuit potential [V]": 0.0})
cycle_parameter_values.update({"SEI kinetic rate constant [m.s-1]": 1 * 1e-12})

cycle_experiment = pybamm.Experiment(
    [
        (
            "Discharge at 0.1 C until 2.0 V",
            "Rest for 30 minutes"
        )
    ] +
    [
        (
            "Charge at 1 C until 3.6 V",
            "Discharge at 1 C until 2.0 V",
        )
    ] * 1000 + 
    [
        (
            "Charge at 0.1 C until 3.6 V",
            "Discharge at 0.1 C until 2.0 V",
            "Rest for 10 minutes"
        )
    ] +
    [
        (
            "Charge at 0.1 C until 3.6 V",
            "Hold at 3.6 V until 50mA",
            "Discharge at 0.1 C until 2.0 V"
        )
    ]
)

cycle_sim = pybamm.Simulation(cycle_model, cycle_experiment, parameter_values=cycle_parameter_values)
cycle_sim.solve(solver=pybamm.CasadiSolver(mode="safe"))
cycle_solution = cycle_sim.solution.cycles[-1]

ref_cap_Wh = get_Wh(ref_solution)
print(f"Capacity after reference experiment: {ref_cap_Wh} Wh")

cycle_cap_Wh = get_Wh(cycle_solution)
print(f"Capacity after experiment: {cycle_cap_Wh} Wh")

soh = cycle_cap_Wh / ref_cap_Wh * 100
print(f'SOH: {soh} %')
```

### Relevant log output

```shell
raise pybamm.SolverError(
pybamm.expression_tree.exceptions.SolverError: Could not determine which event was triggered (possibly due to NaNs)
```

tinosulzer, 2022-07-26:
Have found the bug, it is because the voltage (positive electrode OCP) changes so quickly near full charge so it is struggling to find the event where minimum voltage is triggered. Now there is another bug ...

tinosulzer, 2022-07-26:
Should be fixed by #2190 . You'll have to do `calc_esoh=False` in `.solve()` for now, but that should be fixed soon in a separate PR.

kratman, 2024-11-21:
The code above works with correct SEI parameters

closed 2024-11-21 by kratman