pybamm-team/PyBaMM #2168: [Bug]: Using `build=False` when creating model doesn't work

brosaplanella, 2022-07-11:
### PyBaMM Version

22.6

### Python Version

3.7

### Describe the bug

In version 22.6 if you create a model with `build=False` the model is built anyway. That didn't happen in 22.5

### Steps to Reproduce

1. Run
```python3
import pybamm

model = pybamm.lithium_ion.SPMe(build=False)
print(model._built)

model.build_model()
```
2. See error
3. Switch to v22.5 and observe there's no error

rtimms, 2022-07-11:
Is it just for SPMe? I think we need to pass `build` [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/spme.py#L47).

brosaplanella, 2022-07-11:
Just checked and it doesn't happen with DFN, good catch! Is there an open PR for it or shall I open it?

rtimms, 2022-07-11:
Great! I am working on #2056 so can fix it as part of that. Hopefully will make a PR today or tomorrow.

brosaplanella, 2022-07-11:
Cool! I can work with v22.5 for the moment so no rush :)

rtimms, 2022-07-12:
Fixed by #2169

closed 2022-07-12 by rtimms