pybamm-team/PyBaMM #2188: Introduction of 3 electrode setup

dion-w, 2022-07-24:
### Description

The feature involves additional potential variables. The first one equates to a potential difference between the current collector of the anode and the middle of separator called "Negative electrode potential vs separator reference". The second one is the potential difference between the middle of the separator and the current collector of the positive electrode called "Positive electrode potential vs sparator reference".

### Motivation

Motivation for this comes from the 3-electrode setup in the lab. 

In a 2-electrode setup in the lab one can not resolve the individual SOCs of each electrode. You only get the potential difference between both current collectors. In a 3-electrode setup a reference electrode is introduced in the middle of the separator for example a little place of lithium or a whole lithium reference ring around the separator. With the reference electrode, the potentials of both electrodes can be monitored separately. These potentials can for example be used for plating detection on lithium ion graphite anodes.

Modelling this feature directly in PyBaMM is not yet possible. There are already a lot of fancy ways in pybamm to resolve processes on the electrodes directly for example different overpotential contributions. These though can not be compared directly to measured data in the lab. With the mentioned feature there is a way to describe these electrode potentials in PyBaMM and relate them directly to measured potentials in the lab.  The additional potential variables could help a great deal for fitting models better to measurements and increase model validity.

### Possible Implementation

As a rough draft for my own simulations i implemented the following lines in "base_electrolyte_conductivity". It could be placed in the current-collector submodel or maybe also "base electrode" but since the separator electrolyte potential is needed this would involve including a new submodel at these places. For now these lines worked for me:

```
        phi_e_s_mid = pybamm.x_average(phi_e_s)  # CAVEAT
        phi_e_s_mid_dim = -param.U_n_ref + phi_e_s_mid * param.potential_scale
        v_n = pybamm.boundary_value(variables["Negative electrode potential [V]"], 'left') \
              - phi_e_s_mid_dim
        v_p = pybamm.boundary_value(variables["Positive electrode potential [V]"], 'right') \
              - phi_e_s_mid_dim
        variables.update(
            {
                "Negative electrode potential vs reference [V]": v_n,
                "Positive electrode potential vs reference [V]": v_p,
            }
        )
```
There is one caveat though. For a point of reference we need a single value (here termed "phi_e_s_mid"). Typically this would just be the evaluation of the electrolyte in the middle of the separator but until now i did not manage to get it working. I tried to do it with `phi_e_s.evaluate(param.L_n+param.L_s/2)' but it didn't want to be evaluated as it said the method "evaluate is not impleted for "Current function"'. I did not try it with another approach as this wouldn't be working for my system.

### Additional context

Provided there is feedback and discussion and it's not too much work to put the feature inside while making sure that everything else in the code still works i would implement this and create a pull request.

brosaplanella, 2022-07-24:
I had a similar issue not that long ago in terms of where are the three electrode potentials defined from. Is it consistent across cells or it depends on the implementation? If the latter, it might be better to leave it to the user to define the variable. Otherwise, the average might be a good approach (as a good approximation).

tinosulzer, 2022-07-25:
Yeah I agree with Ferran, it all depends how precisely the separator reference is positioned. My guess is it doesn't need to be very precise, since the electrolyte potential doesn't change much over the separator, and so `x_average(phi_e_s)` is a good approximation. If it is precise, we could implement a new symbol like `EvaluateAt(phi_e_s, x_s=...)`. `.evaluate()` doesn't (isn't intended to) work on some `Symbol` types, e.g. `Parameter` objects that don't have a value.

dion-w, 2022-07-26:
Referring to @tinosulzer, this was exactly my orginal idea with `EvaluateAt`. For now the `x_average` method only gets the value in the exact middle if and only if the separator electrolyte potential is linear in this region (which it is in the P2D). To make it more robust for other circumstances, future releases and model extension the symbols `EvaluateAt` would be helpful since this method would come with no such assumption. 
So referring to a possible implementation with `EvaluateAt`: What do you think if we add another parameter in the "cell"-File (e.g. called "Point of reference potential, normalised distance to negative electrode")  that describes the normalised distance from the right boundary of the negative electrode to the left boundary of the right electrode in the separator region? A 0.5 would be right in the middle. This way there could be stable implementation and the user could even choose where he wants to have his reference potential.

tinosulzer, 2022-07-29:
Yes, adding that parameter is a good way to do it. Out of interest, how much control is there over the position of the reference electrode in experiments?

dion-w, 2022-07-30:
> Yes, adding that parameter is a good way to do it. Out of interest, how much control is there over the position of the reference electrode in experiments?

Yeah that's the point what Ferran talked about. I know of two particular experimental setups where you're testing electrodes in the lab in the coin cell format. One is a setup with so called "[swagelok-cells](https://doi.org/10.1002/celc.201500474)" (swagelok coming from the company offering these), and another setup sold commercially is from the company EL-CELL with so called [PAT-cells](https://el-cell.com/pat-series/the-pat-core-concept/). In both circumstances the reference electrode sits exactly in the middle of the separator. This way a constant offset potential for both electrodes is created and so they are comparable in terms of potential difference. With PAT-Cells there is even a ring electrode around the separator to increase validity of impedance spectra measured with the setup. 
So this is for lab scale coin cells. It may be different though if you want to introduce a reference into, let's say, pouch cells but i never worked with these.

So basically: the position of the reference electrode in the electrode is mostly set in stone for experimental coin cell setups and there is actually no need to introduce a movable reference electrode in pybamm. I just mentioned the adjustable distance because of Ferran. My suggestion would be to either set it to 0.5 by default, or don't use the additional parameter and just implement it hardcoded to 0.5 in the source code. 
but again, while i did some stuff in the lab, i'm using 90 % of my time for simulations, so it might useful to consider a full time experimentalists for this question.

dion-w, 2022-08-02:
Also found a paper nicely explaining the ins and outs of the three electrode setup and what's generally important when dealing with reference electrodes.

tinosulzer, 2022-08-02:
@dion-w are you wanting to get started on implementing this? I can give you some pointers if so

dion-w, 2022-09-02:
@tinosulzer You can give them to me. If nothing happened after two weeks i likely had too much to do. But i can try for now as i'm still on vacation.

tinosulzer, 2022-09-02:
First step will be to implement the `EvaluateAt` as part of the expression tree. I would recommend doing this in small pieces, one step per PR. 
1. add new `EvaluateAt` class to `UnaryOperators` (subclass `SpatialOperator`), add tests that it works as expected. This should act pretty similarly to `BoundaryOperator` in terms of how it affects domains, so you can copy that [class](https://github.com/pybamm-team/PyBaMM/blob/1f01c1e391cc7d01feba405e40584e2f6c456fae/pybamm/expression_tree/unary_operators.py#L865) and [tests](https://github.com/pybamm-team/PyBaMM/blob/1f01c1e391cc7d01feba405e40584e2f6c456fae/tests/unit/test_expression_tree/test_unary_operators.py#L478) to get started
2. implement the finite volumes discretization for `EvaluateAt`
3. Add an model option for "three electrode setup", if it's "true" then add the reference potential as a variable in `base_electrolyte.py` (we don't want to do this if it's "false" so that the parameter for the location of the reference electrode isn't required by default)

dion-w, 2022-09-27:
Sorry, i'm busy now and can't say when i would have time for this. You can give it to someone else or just keep it one the list for future use.

mpegis, 2023-05-17:
This is pretty easy to do assuming you have a Li-sensitive reference and know its stable in standard potential vs Li. I saw rob open this and thought I would give my two cents.

First off, assuming you have an ideal experimental cell where the reference plane is perpendicular and equipotential through the length and width of the cell, and you are using the reference electrode as an auxiliary voltage measurement (passive reference). All you basically need to do is calculate the electrolyte potential at that distance, and subtract it from the positive/negative electrode potentials. Its essentially just a thermochemical cycle that can be accessed directly from the output solution, since electrolyte potential is vs Li as a default setting.

L_n = parameter_values["Negative electrode thickness [m]"]
L_s = parameter_values["Separator thickness [m]"]
L_p = parameter_values["Positive electrode thickness [m]"]
x_mid = L_n + L_s / 2
x_anode = L_n 
x_cathode = L_n + L_s
V_ref_mid = sols[i]["Electrolyte potential [V]"](t=t, x=x_mid)
V_ref_A = sols[i]["Electrolyte potential [V]"](t=t, x=x_anode)
V_ref_C = sols[i]["Electrolyte potential [V]"](t=t, x=x_cathode)
Anode_V_A = sols[i]["Negative electrode potential [V]"](t=t,x=0) - V_ref_A
Anode_V_C = sols[i]["Negative electrode potential [V]"](t=t,x=0) - V_ref_C

The real challenge I think is going to be
1. Actually getting an equipotential reference. The PAT cells actually have a ring located on the bottom end of the separator in the core, kind of displaced outside the anode/cathode dimensions. Its unclear to many how the distributed ohmic drop across the electrolyte with this kind of radial symmetry is going to impact your results. They do make finger shaped references now, but its always a trade-off (you have to sandwich it between two thick separators.

2. Assuming you feel confident your setup has solved issue #1, its going to be more challenging to actually simulate an active reference, where E control/limits are set vs the ref, not the counter. I suppose one could modify the approach to the solution and get some sort of feedback cycle where it will do a time-step and find an acceptable solution state, compute the active V control limit based on geometry of the ref and continue. I don't know enough about the architecture to make a valid opinion regarding this step.

Just my two cents. hope its helpful

PR #3573 (2023-12-05): #2188 implement evaluate at in 1D
# Description
Implement a new unary operataor `EvaluateAt` that evaluates a spatial variable at a given position. For now, this is only implemented for 1D spatial variables using the Finite Volume method. In this method, `EvalauteAt` returns the variable evaluated at the node closest to the requested value. E.g. `EvaluateAt(3*x, 1.9)` on a mesh with nodes`x=[0.5, 2.5, 4.5]` would return 7.5.

The motivation for this is to be able to place a reference electrode at a given point in the cell, to mimic a 3E setup.

Fixes #2188 

## Type of change

Please add a lin [... cut]
closed 2023-12-05 by rtimms