pybamm-team/PyBaMM #2193: [Bug]: Wrong starting_solution for cycle-experiments

MichaPhilipp, 2022-07-27:
### PyBaMM Version

22.5

### Python Version

3.9.12

### Describe the bug

After initialising a cycle-simulation, solving it multiple times in a row leads to wrong starting_solution/initial values for the 2nd/3rd/... solution. In detail this occured by simulating a cycle-experiment for SEI-Growth by varying SEI-parameters for each simulation. The very first simulation contains the correct intial values, however the second simulation starts with a SEI-thickness close to the final solution of the previous simulation. This continues for each following simulation.

I think it is supposed to start with the same initial inner/outer SEI-thickness for each simulation. Note that the starting SEI-thickness of the (i+1)-th simulation is not exactly the final solution of the i-th simulation, but always a bit below that value. I`ve attached a MWE below for this behaviour, it is straightforward to reprocude.
Is this how it is supposed to work, or how can I set the starting_solution to the initial values of the simulation without re-initialising the simulation again?

### Steps to Reproduce

`
import pybamm
import matplotlib.pyplot as plt
import numpy as np
from pybamm.models.full_battery_models.lithium_ion.electrode_soh import get_initial_stoichiometries
from copy import deepcopy

### MWE ###
seigrowth= "interstitial-diffusion limited"

model = pybamm.lithium_ion.SPMe({"SEI": seigrowth})

parameter_values = model.default_parameter_values
param = pybamm.lithium_ion.SPMe().param

c_n_max = parameter_values.evaluate(param.n.c_max)
c_p_max = parameter_values.evaluate(param.p.c_max)

V = 3.9
Vmin= 3.0
Vmax= 4.1

experiment = pybamm.Experiment(
    [ "Discharge at 0.1C until 3 V",
      f"Charge at 0.1C until {V}V",
     "Discharge at 0C for 10 hours",
     ]*4,)

parameter_values.update({'Lower voltage cut-off [V]': Vmin,'Upper voltage cut-off [V]': Vmax,})
parameter_values.update({'Initial inner SEI thickness [m]': 1e-12,
                            'Initial outer SEI thickness [m]': 1e-12,})
initial_soc = 0.9
x,y = get_initial_stoichiometries(initial_soc=initial_soc, parameter_values=parameter_values)

parameter_values.update({
    "Initial concentration in negative electrode [mol.m-3]": x * c_n_max,
    "Initial concentration in positive electrode [mol.m-3]": y * c_p_max,})

free_parameter0={'Inner SEI lithium interstitial diffusivity [m2.s-1]': 1e-16}
free_parameter1={'Inner SEI lithium interstitial diffusivity [m2.s-1]': 2e-16}
free_parameter2={'Inner SEI lithium interstitial diffusivity [m2.s-1]': 3e-16}
free_parameter3={'Inner SEI lithium interstitial diffusivity [m2.s-1]': 4e-16}
free_parameter_values=[free_parameter0,free_parameter1,free_parameter2,free_parameter3]
parameter_values.update({name: "[input]" for name in free_parameter0})

solver = pybamm.Simulation(model, experiment=experiment, parameter_values=parameter_values)

for i in range(0,4):
    solution = solver.solve(inputs=deepcopy(free_parameter_values[i]))
    plt.plot(solution.t,solution["X-averaged total SEI thickness [m]"].entries, label=model.name, color="red")

plt.show()
`

tinosulzer, 2022-07-29:
@all-contributors add @MichaPhilipp for bugs

tinosulzer, 2022-07-29:
Should be fixed by #2204, but it's unlikely to make it into `v22.7` so will have to wait until `v22.8`

tinosulzer, 2022-08-02:
22.7 was delayed, will be fixed in 22.7

PR #2204 (2022-08-02): Issue 2193 starting solution
# Description

Fixes bug with starting solution, which gave different solutions when running an experiment multiple times

(can wait until the next release, will update changelog after)

Fixes #2193 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change [... cut]
closed 2022-08-02 by tinosulzer