pybamm-team/PyBaMM #2206: Heat source terms by domain

rtimms, 2022-08-01:
### Description

Add variables e.g. "Negative electrode Ohmic heating [W.m-3]".

Note: depending on which submodel is used, "Electrolyte current density" isn't always a concatenation, so can't be split by domain. Should be able to change this though?

rtimms, 2024-11-22:
tagging @aabills

aabills, 2024-11-22:
The issue with electrolyte current density still stands, so I don't think we can close this. 

But, it should be closed by the second part of our intended submodel refactor, which should change the electrolyte current density.