pybamm-team/PyBaMM #2214: Multiple step conditionals in PyBaMM Experiment description.

itsjacobhere, 2022-08-03:
### Description

Consider the following experiment:

```
exp1 = pybamm.Experiment([
        ("Discharge at  1C for 30 min", 
         "Charge at 0.5C until 4.2 V",
         "Hold at 4.2 V for X seconds"
        ),
    ])
```

Multi-step conditionals would allow one to place constraints over portions of a scenario. For example, if you wanted to constrain the last two steps in `_exp1_ ` to a total time of 60 minutes. In a case like this `X seconds` in the last step, depends on the length of time taken to complete the previous step. 

### Motivation

This would allow for more complex experiments/scenarios increasing the versatility of the Experiment class.

### Possible Implementation

I would suggest adding the ability to specify multi-step conditions, however there may be a more efficient way to do this.

tinosulzer, 2022-08-03:
Can you suggest syntax ?

tinosulzer, 2022-08-03:
#1839 is related

itsjacobhere, 2022-08-03:
hmm maybe something like the following:

`exp1 = pybamm.Experiment([
        ("Discharge at  1C for 30 min", 
         "Charge at 0.5C until 4.2 V == t seconds",
         "Hold at 4.2 V for 3600 - t seconds ",
        ),
    ])`
or if there was a way to store metrics regarding one step and use it in another? 

Another maybe more simple syntax might be something like:

`exp1 = pybamm.Experiment([
        ("Discharge at  1C for 30 min", 
         "Charge at 0.5C until 4.2 V and hold at 4.2 V for 3600 seconds ",
        ),
    ])`

where this would combine the steps but it would infer that the scenario is to complete the first portion (Charge at 0.5C until 4.2 V ) then hold at 4.2 V until the remaining time is up. 

Thoughts?

tinosulzer, 2023-05-08:
This is reasonable, but duplicate of #1839

closed 2023-05-08 by tinosulzer