pybamm-team/PyBaMM #2217: Improvements to IDAKLU solver

martinjrobins, 2022-08-05:
### Description

Many things still left to do on the IDAKLU solver, here is the copied list from #1863 

- [ ] we're allocating memory for return vectors in the solve, this could be done in solver setup
- [ ] we initialise ida_mem and everything every time, could do this once when solver is created, or only on the 1st solve etc
- [ ] create an options dict to pass to the solver, e.g. option to print solver stats,  don't/do use jacobian
- [ ] add option to use iterative solvers
- [ ] add option to use other linear solvers in ida
- [ ] add option to use dense jacobian
- [ ] benchmarks for all of the above solver options
- [ ] investigate if we need to densify the rhs_algebraic casadi evaluation, could it return a sparse vector?
- [ ] investigate if constructing a new casadi function for the residual (rather than doing it manually in c++) is faster
- [ ] investigate if constructing a new casadi function for the sensitivity calculations (rather than doing it manually in c++) is faster
- [ ] we calculate consistent initial conditions at the start of the idklu solve, so we might not need to do it in the BaseSolver setup

### Motivation

1. The IDAKLU solver can be faster than casadi, so we want to expose more functionality from sundials
2. Make the solver more efficient

martinjrobins, 2022-08-08:
"we're allocating memory for return vectors in the solve, this could be done in solver setup"
 Need to think more on this one, if we re-use this memory for subsequent solves will overwrite solution returned to user :(

tinosulzer, 2022-08-08:
There are still scenarios where that's ok - e.g. parameter estimation - so it could be an optional feature?

tinosulzer, 2022-10-12:
Been reading a bit more about Sundials. Super excited to see these changes when they're ready! For pack modeling it could be extremely useful

tinosulzer, 2022-12-13:
Are we now in a place where we can make the IDAKLU solver the default if it's installed (and fall back to casadi otherwise)?
Would be interesting to test/benchmark

martinjrobins, 2023-01-10:
Re benchmarks, it looks like the main advantage of the idaklu solver is for running experiments:

```
 ============ ============= ======================================================= =========================================== ===========================================
              --                                                                                                                       solver_class                                     
              ---------------------------------------------------------------------------------- ---------------------------------------------------------------------------------------
               experiment    parameter                         model_class                        pybamm.solvers.casadi_solver.CasadiSolver   pybamm.solvers.idaklu_solver.IDAKLUSolver 
              ============ ============= ======================================================= =========================================== ===========================================
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM                    876±3ms                                     798±6ms                  
                  CCCV      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN                   3.06±0.01s                                  2.05±0.01s                
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM                    1.31±0s                                     638±2ms                  
                  CCCV        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN                   6.84±0.01s                                  1.78±0.01s                
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.spm.SPM                    875±2ms                                     642±1ms                  
                  GITT      Marquis2019   pybamm.models.full_battery_models.lithium_ion.dfn.DFN                    7.19±0s                                     2.65±0s                  
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.spm.SPM                    1.45±0s                                     548±2ms                  
                  GITT        Chen2020    pybamm.models.full_battery_models.lithium_ion.dfn.DFN                   18.0±0.01s                                  2.62±0.01s                
              ============ ============= ======================================================= =========================================== ===========================================
```

closed 2024-05-14 by martinjrobins