pybamm-team/PyBaMM #2219: Calculate `initial_soc` from model states

brosaplanella, 2022-08-08:
### Description

If I understand correctly the `initial_soc` is initialised from the parameter values. It would be good to initialise it from the model states (if the model initialised from solution).

### Motivation

This would allow for experiments such as RPT, in which we want to start from a given state (e.g. ageing) but run the simulation from 100% SoC.

tinosulzer, 2022-08-08:
Good idea, do you know how to do this?

brosaplanella, 2022-08-09:
> Good idea, do you know how to do this?

Not sure haha. To calculate the right stoichiometry, it should be enough to call `ElectrodeSoH` with the averaged concentrations in each electrode. However, I need to look how to then pass the new initial conditions. If you have any ideas or suggestions let me know.

rtimms, 2023-03-06:
# Hackathon info
This will be one of the projects for the first PyBaMM Hackathon. Here you can find some information (summarising other points made on this issue) which should be a useful starting point. Note that this post will evolve, even during the Hackathon day, so make sure to check it regularly.

## Background
Currently, you can pass `initial_soc` to a `Simulation` to set the initial state of charge. When you do this the simulation calls `set_initial_soc` (defined [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/simulation.py#L405)), which in turn ends up calling `get_initial_stoichiometries` (defined [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py#L509)). This uses the `ElectrodeSOH` algorithm to find the stoichiometry limits based on the parameter values using the algorithm from [Mohtat, 2019](https://pdf.sciencedirectassets.com/271367/1-s2.0-S0378775319X00142/1-s2.0-S0378775319303593/am.pdf?X-Amz-Security-Token=IQoJb3JpZ2luX2VjEFoaCXVzLWVhc3QtMSJHMEUCIHnnfhqE4Y0mmvshhP7N3hd78FCCjhQA6zTispMV9xATAiEAgPC%2B0PMiSwb8yPZrbvrxP4dSNVLbQo1PREUVDlj86SgqswUIEhAFGgwwNTkwMDM1NDY4NjUiDAcIARwcDq1sHPf8oyqQBYv38D2pWRGA84q3prjmoSV103au5Izns6xn5LfAwBxHH6zsn4fJNgMpBPAI3wWkKg81ZKmU%2BRrTrDygGJZniwWTl3M%2BCY1gdYZAp1plosas73gmiccTjVK0ECfL7Yg2asCjeXy8GUq8Cm4BqGFULG%2BEcaclsGCiItP2l2cdMtKcj9y2Z0nocpkmtTIJSOgyb7%2BnSZse2r%2FNHxvHkoMWtHgofh72zxQknqXC%2Fmh3EAHewdVBOTkhDtUSjQvpadZBNxZfpVQwqtu9zGnqz8j0Ld3oyw627Da1kay%2FWdJ0P8egz0QBPFlxJXTG3ssTS7gK1WYZ0apjiiUs%2BLf%2BtvynVE7ITU65XnzxRy%2FwHXJ9Tu4%2BEKA%2FqerLPG11DsVarw8a20iHhvxgdXVGlgGvRNYel10c0OoDS0iVs5pV79qmFAE10R%2Fvq6ioh0O8ESExM92sdlxtTeXviHNZqNw4WTQxu1vyyZ6%2FJgMPmnd2wNKYyERsZ8ghuxKacCu6HJ37EUj8WZGji6c8vqTYiIP6F4tTH8d74NKqTDefdslwQjRneMNlHN%2B%2FFmtITrxY78jBnLAfvaa7foX7XLgeLJZ%2FUVs5DnxTgEy1Y1s%2FRDhKpWbCUQ1pMu2%2FNS16ysj8GWfFn49hLX1pM7izjkVybQlTFE%2F8uXgXmPZgakN2HLtjIfw7TX7banIC9X%2BugsfK2mzDdG3%2BFScBQ4t2gNfxj%2FB3ZfSoPLOhPayVWaxWOJYDCd2bd%2F9TeX1Lg6hRZRacNJEHqTqBiPwl1gYE70W2cDXZz2Nc66MAef38UrTN8tBMy8i3gMc%2BP8ePGdocAD2%2BXZugoMFYmNWMSUJBE7AICQME6LYn16H4nD34LotoOw4YO6jxSFu%2FMKnglqAGOrEB%2BS6oPeX%2FCDg4%2BucKNjLr1W71s92f9ljy6xgwbGa8EiZHNGjQevy2XOC9KRnnkULprsWagat13t%2BzuZ6OtQL8iCM7WiLfpHJmxchb4%2F%2F2OCSUJzpSLdnesre2y%2BZdOLn8mpZhPt3BRiOup0f2doE4XTc6HlMI7R69VXaAl2%2Bp8lgU%2Bu7sENJpERBIGQlXimWUGegWiQnLOTo6WVge2hzteKygr%2BHqMzDRZ7%2F0Z1SL0KPa&X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Date=20230306T102915Z&X-Amz-SignedHeaders=host&X-Amz-Expires=300&X-Amz-Credential=ASIAQ3PHCVTYYTYJBK4N%2F20230306%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Signature=0a4755ae34808dc634b8c5a922c6588dac4867885cfcfc158e66763bd6696a9a&hash=80d1b83ae763f369e02247546df9d106b0d647cc7d366798e1758d03c4eb8679&host=68042c943591013ac2b2430a89b270f6af2c76d8dfd086a07176afe7c76c2c61&pii=S0378775319303593&tid=pdf-5fc464e3-f978-4de6-96ca-1edae31d2e1b&sid=5313c3ec4007c143e75ae1c2b77e93f02dacgxrqb&type=client).

You can also pass a `starting_solution` to a `Simulation` to start a new simulation using an old solution as an initial condition. It would be good to be able to set a new initial state of charge using the states from the `starting_solution` instead of the `parameter_values` - this would make it so the initial soc calculation could be made from an aged state.

## Tasks
- Rewrite `get_initial_stoichiometries` so you can pass a `starting solution` to use in the calculation. This will require changing a bunch of the [electrode state of health code](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py). In particular, the electrode capacities and total amount of lithium used [in this calculation](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/full_battery_models/lithium_ion/electrode_soh.py#L482) for the min/max stoichiometries will need to be (optionally) calculated from the model states instead of the parameter values.
- Write an example that manually calls the updated `get_initial_stoichiometries` using the solution of a simulation to get the new stoichiometries. Manually update the initial conditions and use this to create a new simulation.
- Based on the example, update `set_initial_soc` in `Simulation` to behave appropriately depending on whether a `starting_solution` is passed or not.

rtimms, 2023-03-30:
Having attempted this at the hackathon, it is unclear if it is a sensible thing to do. If you stop a simulation at non-equilibrium and instantaneously change the SOC, does that make sense? Probably not. 

We have a work-in-progress notebook showing how to set up an experiment where you, for example, do 1000 cycles of a CCCV experiment, but stop every 100 cycles and do a slow charge followed by an RPT. Then we show how to post-process all this to get informative plots. This seems to be the best solution and might inform the classes introduced as part of #2322.

FYI, if you calculate "summary variables" as part of the simulation then you can get the new min/max stoichiometries from the summary variables. If you wanted to reach a given SOC for the aged cell you could use the new min/max stoichiometries along with the OCV and target SOC to figure out a target voltage to charge to.

I propose we close this once the new example notebook is merged.

brosaplanella, 2023-03-30:
Makes sense! Sorry I set you for trouble with this issue hahah

PR #2851 (2023-10-03): Issue 2219 initial soc from experiment
# Description

Added new rpt-experiment notebook to demonstrate how to set up degradation experiments with reference performance tests (RPTs).

Fixes #2219 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] B [... cut]
closed 2023-10-03 by brosaplanella