pybamm-team/PyBaMM #2220: Check coord_sys of SpatialVariable

rtimms, 2022-08-09:
When creating a `SpatialVariable` check the coordinate system against `KNOWN_COORD_SYS` defined [here](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/expression_tree/independent_variable.py#L8).

brosaplanella, 2022-08-09:
You beat me hahaha. The docstring also needs fixing.

rtimms, 2022-08-09:
Also, add a setter for `coord_sys` is `SubMesh` to perform this check for the meshes too.

kratman, 2024-05-03:
Is this solved by #3394 and #3114? This seems like a duplicate

agriyakhetarpal, 2024-05-03:
Yes, Ferran mentioned in https://github.com/pybamm-team/PyBaMM/issues/3114#issuecomment-1648160595 that #3394 should close both issues. We should probably close this one and keep the newer one open.

closed 2024-06-03 by github-actions[bot]