pybamm-team/PyBaMM #2221: BaseBatterySubmodel

rtimms, 2022-08-09:
Make a `BaseBatterySubmodel` that performs things like checking domains, e.g.
```
        ok_domain_list = [
            "Negative",
            "Separator",
            "Positive",
            "Negative electrode",
            "Negative electrolyte",
            "Separator electrolyte",
            "Positive electrode",
            "Positive electrolyte",
            None,
        ]
```
so that `BaseSubmodel` is more general. Useful for training etc. to not be restricted to battery things when using submodels.

tinosulzer, 2022-08-09:
We could consider getting rid of those checks.

brosaplanella, 2022-08-10:
Following our experience with the `coord_sys`, I think we should keep some checks somewhere in the battery submodels. But certainly not in the `BaseSubmodel` class.