pybamm-team/PyBaMM #2225: [Bug]: Particle size distribution concentration plot x axis may be incorrectly labelled or calculated

TomTranter, 2022-08-11:
### PyBaMM Version

22.7

### Python Version

3.9

### Describe the bug

[image]

### Steps to Reproduce

```
import pybamm
model = pybamm.lithium_ion.MPM()
sim = pybamm.Simulation(model)
sim.solve([0, 3600])
sim.plot([
    "X-averaged negative particle concentration distribution",
    "X-averaged positive particle concentration distribution",
    "Terminal voltage [V]",
])
```

iatzak, 2023-02-06:
In [line 365 of the `ProcessedVariable` class](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/solvers/processed_variable.py#L365), there is a conditional regarding the `first_length_scale` of such r-R plots, hard-coding it to 1. I couldn't find out where r is scaled with R, as the comment there says.

Removing the conditional causes the value `1e-5` to get assigned to `first_length_scale` for the r-R plots of your example, giving

[image]

Do you think this is the way to go? Apart from that, could you assign me to this issue?

tinosulzer, 2023-02-06:
Yes this looks like the way to go, thanks!

PR #2663 (2023-02-13): Remove conditional for scaling of r-R plots
# Description

Removing hard-coded value of 1 for `first_length_scale` of r-R plots, which made the x axis of these plots be scaled by 1e5.

Fixes #2225 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] [... cut]
closed 2023-02-13 by tinosulzer