pybamm-team/PyBaMM #2226: [Bug]: Equations not displaying properly in docs for 1D submesh

TomTranter, 2022-08-11:
### PyBaMM Version

22.7

### Python Version

3.9

### Describe the bug

https://pybamm.readthedocs.io/en/latest/source/meshes/one_dimensional_submeshes.html

### Steps to Reproduce

https://pybamm.readthedocs.io/en/latest/source/meshes/one_dimensional_submeshes.html

eagunn, 2023-01-06:
Contributing a comment rather than a fix. I'm not set up to build and test. I was just cruising through the issues deciding if this was a project where I might be able to make contributions in the future. But, since I think I've figured out what is wrong with the PyDoc comments, I thought I would leave a note here to possibly make the eventual fix faster for the next in line. 

I looks as if the documentation for these classes is stuck between two approaches to documenting exponents of e. See [https://tex.stackexchange.com/questions/254785/e-vs-exp-in-display-mode](https://tex.stackexchange.com/questions/254785/e-vs-exp-in-display-mode) for a helpful discussion of the two approaches.

Specifically, I believe the raw equation in the Pydoc:
`x_{k} = (b-a) + \\frac{\\exp{\\alpha k / N} - 1}{\\exp{\\alpha} - 1} + a,`
should be more like
`x_{k} = (b-a) + \\frac{\\exp(\\alpha k / N) - 1}{\\exp(\\alpha} - 1) + a,`

That is, the Tex operator \exp is to be followed by the expression in parentheses, not curly brackets.

dikwickley, 2023-01-26:
@tinosulzer @TomTranter i would like to work on this issue.

tinosulzer, 2023-01-26:
Let us know if you need help

dikwickley, 2023-01-27:
are these the equations that are not being displayed correctly?
[image]
from what I can understand, the exponent part is displayed incorrectly.

and if yes, can you point me toward any source material that shows how they should actually look like?

dikwickley, 2023-01-27:
@tinosulzer i have created a PR, please review it. thanks

PR #2627 (2023-02-17): fixed exponent in doc string submesh 1d
# Description

Changed the equation string in submesh 1d DOC string to show proper exponent

Fixes #2226 

before:
[image]

after:
[image]

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [ ] No style issues: `$ flake8`
- [ ] All tests pass: `$ python run-tests.py --unit`
- [ ] The documentation builds: `$ cd docs` and then `$ make clean; make html`

You can run all three at once, using `$ python run- [... cut]
closed 2023-02-17 by tinosulzer