pybamm-team/PyBaMM #2232: Loss of capacity to loss of active material

frank4399, 2022-08-15:
### Description

According to Birkl, Christoph R., et al. "Degradation diagnostics for lithium ion cells." Journal of Power Sources 341 (2017): 373-386., loss of active material on either positive electrode or negative electrode may have an impact on capacity fade. Is this feature going to be implemented and included in solution.summary_variables?

tinosulzer, 2022-08-16:
It is already implemented both as a model option (https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/models/submodel_loss_of_active_materials.ipynb) and in summary variables.

frank4399, 2022-08-16:
> It is already implemented both as a model option (https://github.com/pybamm-team/PyBaMM/blob/develop/examples/notebooks/models/submodel_loss_of_active_materials.ipynb) and in summary variables.

In the summary variables of a solution, only "loss of active material in negative electrode [%]" is found. I was wondering whether a summary variable like "Loss of capacity to active material in negative electrode [A.h]" would be implemented in the future.

tinosulzer, 2022-08-16:
Ok, you can do this easily yourself. Add
```python
{"Loss of active material in negative electrode [A.h]": param.n.cap_init - C_n}
```
to [this function](https://github.com/pybamm-team/PyBaMM/blob/0bb70d240841225b785c70cdcd6bb7575b1d33ca/pybamm/models/full_battery_models/lithium_ion/base_lithium_ion_model.py#L115)

and then add `"Loss of active material in negative electrode [A.h]"` to [this list](https://github.com/pybamm-team/PyBaMM/blob/0bb70d240841225b785c70cdcd6bb7575b1d33ca/pybamm/models/full_battery_models/lithium_ion/base_lithium_ion_model.py#L187)

and same for positive electrode

frank4399, 2022-08-16:
Understood it now. Thanks a lot for the help, Valentin!

closed 2022-08-16 by frank4399