pybamm-team/PyBaMM #2253: Max step size option

rtimms, 2022-08-23:
Add a `max_step` parameter to `BaseSolver` and pass it to the solvers in the appropriate way. Then for drive cycle simulations we should set `max_step` to the step size in the data.

rtimms, 2022-08-23:
Or at least do that in our drive cycle examples

jaskiratsingh2000, 2023-02-20:
Hey @rtimms What are you excatly expecting out of this? Where can I find this? Thanks!

rtimms, 2023-02-20:
I wanted to add a `max_step` parameter to `BaseSolver` which controls the maximum step size the solver can take. You'll need to look at the docs for the each of the solvers we use as they may all have different interfaces. E.g. for [scipy integrate](https://docs.scipy.org/doc/scipy/reference/generated/scipy.integrate.solve_ivp.html) you can directly pass `max_step` as an extra option. We should probably follow their syntax so that the default is `inf`, i.e., the step size is not bounded and determined solely by the solver.

Note this is different from `dt_max` used in the `CasadiSolver` to set the step size for event checking. I propose we rename `dt_max` to `dt_event`.

This might be more like medium rather than easy.

jaskiratsingh2000, 2023-02-20:
I see. Can you provide a basic example for the same because I'm not getting where other parameters are gettiing called in https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/models/submodels/base_submodel.py

Saransh-cpp, 2023-02-20:
I think you are looking at the wrong class (based on the link you attached). `BaseSolver` -

https://github.com/pybamm-team/PyBaMM/blob/6ab6c15979ad1b11fcebea37e43723179d201ea6/pybamm/solvers/base_solver.py#L19

jaskiratsingh2000, 2023-02-23:
@rtimms Any update on this?

rtimms, 2023-02-23:
What do you need an update on? I think you were looking at the wrong class, as suggested by @Saransh-cpp .

jaskiratsingh2000, 2023-02-24:
@rtimms I was expecting the example on above. That would be helpful if you can provide it.

arjxn-py, 2023-03-07:
Hi there @rtimms I had gone through the [scipy integrate](https://docs.scipy.org/doc/scipy/reference/generated/scipy.integrate.solve_ivp.html) to try to understand what exacty could be done, but I don't find any parameter like max_step there. Would you like to suggest me anything in regards?

medha-14, 2024-12-10:
Could I work on this issue if no one else is currently assigned to it?

agriyakhetarpal, 2024-12-10:
> Could I work on this issue if no one else is currently assigned to it?

Yes – @arjxn-py, if you're not working on #3106 anymore, could @medha-14 try to take over? @medha-14, you can pull that branch from Arjun's fork so that authorship of the previous work is preserved.

arjxn-py, 2024-12-13:
Hi @medha-14, thank you for extending your interest. Please feel free to take over the PR.
Sorry this one fell through the cracks.