pybamm-team/PyBaMM #2259: Temperature data from running simulation[Bug]:

Sourabh-Jadhav-2, 2022-08-24:
### PyBaMM Version

22.5

### Python Version

3.8

### Describe the bug

How to get temperature feedback from a running simulation? so that we will use for current function to modify future current steps

### Steps to Reproduce

**

tinosulzer, 2022-08-24:
How is this different from #2136 ?

Sourabh-Jadhav-2, 2022-08-24:
I saw one notebook explaining about user defined input current function with time, so I will temperature instead of time only that's why I asked

tinosulzer, 2022-08-24:
You want current to be an explicit function of temperature? Something like `I = 1 + (T-298)/10` or whatever?

Sourabh-Jadhav-2, 2022-08-24:
No, I will use temperature feedback as a indicator to change c rate. Example: cell temperature is going beyond 40 Celsius then c rate will decrease to less than 1C. So that we will maintain cell Temperature within safe limit.

tinosulzer, 2022-08-24:
That sounds like the same thing as you posted in #2136

closed 2022-08-26 by tinosulzer

Jadhav-Sourabh, 2022-10-11:
I solved that problem in different way, I want some suggestions from you expert for the improvement