pybamm-team/PyBaMM #2265: Raise error if updating a parameter that already exists with `check_already_exists=False`

tinosulzer, 2022-08-26:
This will force people to use `check_already_exists=False` sparingly, and help to catch bugs related to misnaming parameters (this is caught if `check_already_exists=True`, but not if it's `False`

tinosulzer, 2022-08-26:
Yes

brosaplanella, 2022-08-26:
I guess if you need to do both you will need to call the command twice, but it is not too bad (especially if it helps catching typos).

jaskiratsingh2000, 2023-02-20:
Hey! Can you let me know that what exactly is the issue. where does it needs to eb implemented?

brosaplanella, 2023-02-20:
Hi! It is not good practice to assign multiple issues to a new user at once. Please choose one issue, and after it has been merged we can assign you another one. This applies to many other issues where multiple users have ask to get them assigned, so please reply in the one you want to tackle first.

ayeankit, 2023-02-20:
Hello @brosaplanella, Please assign this to me. I don't have any assigned task right now.

jaskiratsingh2000, 2023-02-20:
Agreed upon @brosaplanella else, it will be unfair for other contributors. I have seen the comments on other issues so it is suggested that you work on particular and then comment on other. I was about to work on this as I have also mentioned about issue to ask upon but that's okay. keep up the good work

brosaplanella, 2023-02-20:
Basically, the idea is to modify the `update` method in [parameter_values.py](https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/parameter_values.py) to throw the error described above.

jaskiratsingh2000, 2023-02-20:
@brosaplanella Should I work on this?

ayeankit, 2023-02-20:
hey @jaskiratsingh2000 , currently I am working on it.

brosaplanella, 2023-02-20:
> @brosaplanella Should I work on this?

@jaskiratsingh2000 I assigned you #2695 earlier, are you happy tackling that one?

jaskiratsingh2000, 2023-02-20:
Hey Sure!

jaskiratsingh2000, 2023-02-20:
Happy to both

ayeankit, 2023-02-23:
Hello, @brosaplanella, 
https://github.com/pybamm-team/PyBaMM/blob/4f03df423bc63f9b4bd5f7e926c78a2ad1af1f87/pybamm/parameters/parameter_values.py#L313-L319

Is this error is raised for the same case or the other case?

kratman, 2023-09-30:
I added a portion of this change in #3382. Although that PR only logs a warning. I did a partial change due to the number of tests and examples that would need to be updated by the change.

For making it an error to update parameters with **check_already_exists=False**, I think one of the following two options could be good:
1. Change **check_already_exists** to something along the lines of **add_new_parameters**. My reasoning is that **check_already_exists=True** sounds like "make sure you don't overwrite parameters" while **check_already_exists=False** sounds like "Don't check, just create or update". It would be less confusing if the variable name was closer to the intended usage once the error is in place. 
2. Split the update() function into two different functions. There could be an update_parameter() and an add_new_parameter(). This would get rid of **check_already_exists** and avoid confusion.

Of those two options, I would lean towards (2) since each function would have a single purpose and there would be no ambiguity from the name of the variable.

brosaplanella, 2023-10-02:
I agree with what you suggest, and I also think option 2 is clearer.

closed 2024-11-21 by kratman