pybamm-team/PyBaMM #2269: Not able to install pybamm on pycharm

phanisreepada, 2022-08-28:
### PyBaMM Version

pybamm 0.2

### Python Version

python 3

### Describe the bug

Error during installation of the package

import wheel.bdist_wheel as orig
      ModuleNotFoundError: No module named 'wheel'
      [end of output]

### Steps to Reproduce

no idea

Saransh-cpp, 2022-08-28:
`pip install wheel` should work. Also, you should install the latest version of `PyBaMM`, that is `22.7`.

phanisreepada, 2022-08-29:
This is the error im facing. Wheel got installed. When i try installing Pybamm again, i got he error the below.
 
ERROR: Cannot install pybamm==0.1.0, pybamm==0.2.0, pybamm==0.2.1, pybamm==0.2.2, pybamm==0.2.2.post1, pybamm==0.2.2.post2, pybamm==0.2.2.post3, pybamm==0.2.3, pybamm==0.2.3.post1, pybamm==0.2.4, pybamm==0.2.4.post1, pybamm==0.2.4.post2 and pybamm==0.2.4.post3 because these package versions have conflicting dependencies.

Saransh-cpp, 2022-08-31:
Can you try `pip install pybamm==22.7` (preferably in a virtual environment with `wheel` already installed)? Also, please paste the command executed by you along with the complete error.

phanisreepada, 2022-08-31:
(venv) PS S:\Python-PyCharm\Pybamm> python --version
Python 3.10.6

(venv) PS S:\Python-PyCharm\Pybamm> pip install wheel                     
Requirement already satisfied: wheel in s:\python-pycharm\pybamm\venv\lib\site-packages (0.37.1)

(venv) PS S:\Python-PyCharm\Pybamm> pip install pybamm==22.7
ERROR: Ignored the following versions that require a different python version: 0.3.0 Requires-Python >=3.6,<3.9; 0.3.0.post1 Requires-Python >=3.6,<3.9; 0.3.0b0 Requires-Python >=
3.6,<3.9; 0.4.0 Requires-Python >=3.6,<3.10; 21.10 Requires-Python >=3.7,<3.10; 21.11 Requires-Python >=3.7,<3.10; 21.12 Requires-Python >=3.7,<3.10; 21.8.post1 Requires-Python >=
3.6,<3.10; 21.8.post2 Requires-Python >=3.6,<3.10; 21.9 Requires-Python >=3.7,<3.10; 22.1 Requires-Python >=3.7,<3.10; 22.2 Requires-Python >=3.7,<3.10; 22.3 Requires-Python >=3.7,<3.10; 22.4 Requires-Python >=3.7,<3.10; 22.5 Requires-Python >=3.7,<3.10; 22.6 Requires-Python >=3.7,<3.10; 22.7 Requires-Python >=3.7,<3.10; 22.8 Requires-Python >=3.7,<3.10   
ERROR: Could not find a version that satisfies the requirement pybamm==22.7 (from versions: 0.1.0, 0.2.0, 0.2.1, 0.2.2, 0.2.2.post1, 0.2.2.post2, 0.2.2.post3, 0.2.3, 0.2.3.post1, 0.2.4, 0.2.4.post1, 0.2.4.post2, 0.2.4.post3)
ERROR: No matching distribution found for pybamm==22.7

Saransh-cpp, 2022-08-31:
We don't support Python `3.10` right now. `PyBaMM` `22.8` should work with Python `3.9`!

closed 2022-09-01 by phanisreepada