pybamm-team/PyBaMM #2281: Adding feature to output individual electrode expansion

sravanpannala, 2022-09-06:
### Description

The model will output positive and and negative electrode thickness in addition to the overall cell thickness it currently outputs.

### Motivation

Currently, the mechanical model outputs only the overall cell thickness. In many cases, especially while studying degradation, it is useful to observe and analyze individual electrode thicknesses i.e. positive and negative electrode thickness.

### Possible Implementation

In the `_get_mechanical_results(self, variables)` function of `base_mechanics.py` we can add additional code so that we can get individual electrode thicknesses. I've already tried out the implementation and it is very simple.
```
        if self.domain+" electrode thickness change [m]" not in variables:
            electrode_thickness_change = 0
        else:
            electrode_thickness_change = variables[self.domain+" electrode thickness change [m]"]
        electrode_thickness_change += self.param.n_electrodes_parallel * v_change * L0
return {
            self.domain+" electrode thickness change [m]": electrode_thickness_change,
        }
```

tinosulzer, 2022-09-27:
How's this going @sravanpannala ?

tinosulzer, 2022-10-19:
@mbonkile is also interested in this feature

sravanpannala, 2022-10-19:
I'll try to work on it this weekend. Just for confirmation, I've to implement this on a fresh fork, right? And then run all the unit tests?

rtimms, 2023-02-09:
@sravanpannala any update on this? Yes, implement on your own fork and then make a PR.

PR #2419 (2023-03-01): Issue 2418 dimensional
# Description

Makes all models dimensional. Major changes:
- Model equations now specified in dimensional form. To help with this, there are a `scale` and `reference` attributes for variables, which can be set to make sure that variables are order 1. Algebraic equations also have to be multiplied by an appropriate number (e.g. `L_x**2`) to make them order 1 and hence pass the tolerance check.
- Removed dimensionless parameters
- Removed most dimensionless variables
- Removed some parameters that were just for non-dimensionalization (typical current, typical electrolyte  [... cut]
closed 2023-03-01 by tinosulzer