pybamm-team/PyBaMM #2308: Plot Expressio Tree

SyarifulAzis1018, 2022-09-21:
I think its good, if we can plot the expression tree of the model, i don't know about how to plot the expression tree,
thank

tinosulzer, 2022-09-21:
You can do `symbol.render()` for a unicode plot in terminal, or `symbol.visualise(filename)` to make a nicer looking figure

SyarifulAzis1018, 2022-09-21:
I'm so sorry, i try to run `Symbol.render()` and i get this error 

> render() missing 1 required positional argument: 'self'

tinosulzer, 2022-09-22:
Do `your_expression_tree.render()` where `your_expression_tree` is the actual pybamm object you want to render. e.g.
```python3
import pybamm
v = pybamm.Variable("v")
tree = 2 * v
tree.render()
```
prints to terminal
```
*
├── 2.0
└── v
```

closed 2022-09-25 by tinosulzer