pybamm-team/PyBaMM #2316: Effect of Crates on Degradation

JSHUAIFE, 2022-09-23:
I tried to study the effect of Crate on degradation throng CCCV cycles using the SEI growth model in PyBaMM.  It is found that higher crate leads to lower capacity loss rate with the same number of cycles.  I think the simulation result is not reasonable. However, I can not see any obvious error in the code (refer to the attached file). I hope to discuss this with PyBaMM experts and find a way to improve the code. 

[image]

~~~
#Simulating long experiments

import pybamm
import matplotlib.pyplot as plt
import numpy as np

#Simulating long experiments 
Crates=[0.1, 0.2, 0.5, 1.0]
Solutions=[]

parameter_values = pybamm.ParameterValues("Mohtat2020")
parameter_values.update({"SEI kinetic rate constant [m.s-1]": 1e-14})
#spm = pybamm.lithium_ion.SPM({"SEI": "ec reaction limited"})
spm = pybamm.lithium_ion.DFN({"SEI": "ec reaction limited"})

#Using the "Electrode SOH" (eSOH) model, we initialize the concentration in each electrode at 100% State of Charge

#Calculate stoichiometries at 100% SOC
esoh_model = pybamm.lithium_ion.ElectrodeSOH()
esoh_sim = pybamm.Simulation(esoh_model, parameter_values=parameter_values)
param = spm.param

Vmin = 3.0
Vmax = 4.2
Cn = parameter_values.evaluate(param.C_n_init)
Cp = parameter_values.evaluate(param.C_p_init)
n_Li_init = parameter_values.evaluate(param.n_Li_particles_init)

esoh_sol = esoh_sim.solve(
    [0],
    inputs={"V_min": Vmin, "V_max": Vmax, "C_n": Cn, "C_p": Cp, "n_Li": n_Li_init},
)
print(f"Initial negative electrode SOC: {esoh_sol['x_100'].data[0]:.3f}")
print(f"Initial positive electrode SOC: {esoh_sol['y_100'].data[0]:.3f}")

#Update parameter values with initial conditions
c_n_max = parameter_values.evaluate(param.c_n_max)
c_p_max = parameter_values.evaluate(param.c_p_max)
parameter_values.update(
    {
        "Initial concentration in negative electrode [mol.m-3]": esoh_sol["x_100"].data[0] * c_n_max,
        "Initial concentration in positive electrode [mol.m-3]": esoh_sol["y_100"].data[0] * c_p_max,
    }
)

pybamm.set_logging_level("NOTICE")

for crate in Crates: 

      experiment = pybamm.Experiment([
          (f"Discharge at {crate} C until {Vmin}V",
          "Rest for 1 hour",
          f"Charge at {crate} C until {Vmax}V", 
          f"Hold at {Vmax}V until C/50")
      ] * 50,
      # termination="80% capacity"
      )
      sim = pybamm.Simulation(spm, experiment=experiment, parameter_values=parameter_values)
      sol = sim.solve()
      Solutions.append(sol)

#Summary variables
#loop through the solutions to compare output_variables

fig, ax = plt.subplots( figsize=(10,8))

for i,  solution in enumerate(Solutions):
    # plot summary variable v/s cycle number
    ax.plot(
        solution.summary_variables["Cycle number"],
        solution.summary_variables["Measured capacity [A.h]"],
        label=f"Crate= {Crates[i]} C ", 
    )

#label the axes
ax.set_xlabel("Cycle number")
ax.set_ylabel("Measured Capacity [A.h]")
ax.set_xlim([1, solution.summary_variables["Cycle number"][-1]])
ax.legend()

print("End of test")

~~~

tinosulzer, 2022-09-23:
As C-rate increases, time per cycle decreases. The literature generally suggests that SEI growth rate depends more on time (calendar ageing) than C-rate. See https://iopscience.iop.org/article/10.1149/2.0241904jes/pdf and https://iopscience.iop.org/article/10.1149/1.3557892/pdf for example.

It would be very interesting to see a more careful analysis of the results:
- SEI growth vs time
- SEI growth vs time, normalizing for C-rate (see fig 7 of the Attia paper)
- SEI growth vs time spent above or below various thresholds of "Terminal voltage [V]" and "X-averaged negative electrode surface potential difference [V]"

For these cases it might be more informative to plot the summary variable "Loss of capacity to SEI [A.h]"

JSHUAIFE, 2022-09-26:
Hi tinosulzer,

Thank you very much for the explanation. 

I plot the growth of SEI layer thickness vs time during different crates with the same number for cycles.

[image]

In addition, I add additional code for simulating the calendar aging and plot the growth of SEI layer thickness with time. 

~~~

# Calculate calendar ageing 
Calendar_Solutions =[]

parameter_values["Current function [A]"] = 0
calendar_sim = pybamm.Simulation(spm, parameter_values=parameter_values)
calendar_solver = pybamm.CasadiSolver(mode="fast")

for i, time in enumerate(Times): 
    seconds=time*60*60
    t_eval = np.linspace(0, seconds, 500)
    calendar_sol = calendar_sim.solve(t_eval=t_eval, solver=calendar_solver)
    Calendar_Solutions.append(calendar_sol)

# loop through the solutions to compare output_variables
fig, axes = plt.subplots(1,1, figsize=(6, 4))

for i,  solution in enumerate(Calendar_Solutions):
    # plot calendar degradation only 
    Time = solution["Time [h]"].entries 
    SeiThickness = solution["X-averaged total SEI thickness [m]"].entries 

    axes.plot(Time, SeiThickness,
        label=f"Period for case of crate= {Crates[i]} C ", 
    )   

axes.set_xlabel("Time [h]")
axes.set_ylabel("SEI Thickness [m]")
axes.legend()

~~~

[image]

Model predictions show that the pure calendar ageing (no current) has the highest growth rate and the higher crate has lower lower growth rate.  The results seem strange to me.

JSHUAIFE, 2022-09-26:
Let's have more discussion on the modelling degradation using PyBaMM!