pybamm-team/PyBaMM #2317: RuntimeError: .../casadi/core/plugin_interface.hpp:417: Plugin 'newton' is not found.

e2sdji, 2022-09-25:
I am a beginner Python battery worker. I encountered this problem today and  can't think of a way to solve it. 
[image]

tinosulzer, 2022-09-25:
Does this work?
```python3
x = MX.sym('x',2); # Two states

# Expression for ODE right-hand side
z = 1-x[1]**2
rhs = vertcat(z*x[0]-x[1],x[0])

ode = {}         # ODE declaration
ode['x']   = x   # states
ode['ode'] = rhs # right-hand side

# Construct a Function that integrates over 4s
F = integrator('F','cvodes',ode,{'tf':4})

# Start from x=[0;1]
res = F(x0=[0,1])

print(res["xf"])

# Sensitivity wrt initial state
res = F(x0=x)
S = Function('S',[x],[jacobian(res["xf"],x)])
print(S([0,1]))
```

e2sdji, 2022-09-25:
Thank you very much for your answer. I just found out the problem,just now. The casadi installed by pip is vulnerable. I downloaded it from the official website to solve the problem.
Thank you !!!!!

tinosulzer, 2022-09-25:
> Thank you very much for your answer. I just found out the problem,just now. The casadi installed by pip is vulnerable. I downloaded it from the official website to solve the problem.
Thank you !!!!!

closed 2022-09-25 by tinosulzer

w1ndfall, 2024-03-31:
I met the same problem, but I found the installation method of casadi on the official website and also used pip. May I ask how did you install casadi on the official website