pybamm-team/PyBaMM #2322: Make "experiment" simulations more customizable

tinosulzer, 2022-09-27:
Add an "intermediate" representation of model+experiment that consists of a list of specialized models to solve, e.g.
```python3
pybamm.Experiment(
    [
        (
            "Discharge at C/5 for 10 hours or until 3.3 V",
            "Rest for 1 hour",
            "Charge at 1 A until 4.1 V",
            "Hold at 4.1 V until 10 mA",
            "Rest for 1 hour",
        ),
    ]
    * 3
)
```
becomes
```python3
models_with_experiment = [
discharge_model,
rest_model,
charge_model,
hold_model,
rest_model
]
```
This is already what `Simulation` does when processing models, but we should make it more accessible so that users can run different types of experiment even if they can't be defined by the `Experiment` class

tinosulzer, 2022-11-21:
@Scottmar93 doing this would make both step-wise temperature control and CC-CT-CV (as we were discussing on slack) very easy, since the hard part is actually processing the strings from experiments

tinosulzer, 2022-11-21:
easy if you know the codebase

tinosulzer, 2022-12-04:
I like the idea and the suggested syntax, a couple of suggestions for syntax changes:
- temperature should be a separate input. Maybe in Kelvin to keep consistent with parameter values.
- drive cycles should be a 2-column numpy array containing the data, rather than a string-reference. It's not too much to ask of the user to load that data from the appropriate file, and saves us having to deal with all the edge cases from different OS
- Cycle ref should be via a `Cycle` object, e.g. `Cycle((Run(...), Rest(...), ref="..."))`
- `until` could be a `StopCondition` class e.g. `StopCondition(V=3.2, T=26)`
- Existing string functionality should still work (e.g. convert to this form automatically) - it's fine if it only works for a subset of cases

See also discussion in #1839 as the start time could be another argument to the experiment step classes

Scottmar93, 2022-12-04:
Really like the ideas in #1839 to help replicate long term usage patterns. Would work well with this context. 

Agree with the drive cycle being passed by the user, makes a lot of sense. Also agree that keeping existing experiment class functionality is a good idea to avoid breaking a bunch of people’s code. Perhaps just doing this in a separate class makes sense? Maybe l call it VirtualCycler or something? 

I lean towards temperature being in degrees Celsius since cyclers use that and batteries are usually kept within -40 to 90oC. I’d be pretty keen to avoid having to convert to kelvin all the time when everything else I deal with is in Celsius but not sure what other use most? 

The other syntax is a trade off between being concise and being explicit. If we have a temperature argument and stopconditon class then writing out each step spreads it over 3 or 4 lines of python code and so becomes a bit harder to see what longer flows are doing. With the shorter syntax, it’s probably easier to read the whole flow but it requires a bit of context to get the details… not sure what is best…

brosaplanella, 2022-12-05:
It might be good to draw inspiration from real cyclers, and how the protocols are passed, so we can add functionality to import those directly into PyBaMM.

Scottmar93, 2022-12-12:
Nice @brosaplanella I'm happy to discuss this when you have some time 😄

Scottmar93, 2023-01-08:
@brosaplanella @tinosulzer 

Was trying out some more versions of syntax for this: compact vs. long-form. Perhaps there is some middle ground if you have any ideas?

Using classes for the cycles, as suggested by Tino, is better. I also think all steps should be contained within a cycle (as this is the case for cyclers I have used). 

**Compact**
```python 
from pybamm import ChargeCCCV, DischargeCC, Rest, Run

model = pybamm.lithium_ion.SPM()

flow = pybamm.Flow(
   "flow name",
    Cycle(
        DischargeCC(at="C/2", until="3.2V", save="1s"),
        ChargeCCCV(at=("C/2", "3.7V", "25oC"), until=("C/50", "5Ah"), ref="CH1"),
        Rest(at="25oC", until="1h 30m", save="1m"),
        Rest(at="35oC", until="3h", save="1m"),
        loop=1
        ref="init"
    ), 
    Cycle(
        Run(at=("US06.csv", "35oC"), until=("3.2V", "0.1*CH1"), save="1s", ref="DRV"),
        Rest(at="35oC", until="1h", save="1s"), 
        loop=4,
        ref="drive cycle"
    ),
    Cycle(
        Rest(at="25oC", until=("5h", "T<26oC"), save="1m"),
        loop=1
        ref="final rest"
    )
)

sim = pybamm.Simulation(model, flow=flow)
```

**Long Form**
```python 
from pybamm import ChargeCCCV, DischargeCC, Rest, Run

model = pybamm.lithium_ion.SPM()

flow = pybamm.Flow(
   "flow name",
    Cycle(
        DischargeCC(
            c_rate=1/2, 
            end_conditions=EndConditions(voltage_v=3.2), 
            save=SaveConditions(time=Time(seconds=1))
         ),
        ChargeCCCV(
            c_rate=1/2,
            voltage_v=3.7, 
            temperature_degc=25, 
            end_conditions=EndConditions(c_rate=1/50, capacity_ah=5), 
            label="CH1"
        ),
        Rest(
            temperature_degc=25,
            end_conditions=EndConditions(time=Time(hours=1, minutes=30))
            save_conditions=SaveConditions(time=Time(minutes=1))
        ),
        Rest(
            temperture_degc=35, 
            end_conditions=EndConditions(time=Time(hours=3)), 
            save_conditions=SaveConditions(time=Time(minutes=1))
        ),
        run=1
        label="init"
    ), 
    Cycle(
        Run(
            filename="US06.csv",
            temperature_degc=35,
            end_conditions=EndConditions(
                voltage_v=3.2, 
                step_capacity=StepCapacity(percentage=10, ref="CH1")
            ),
            save_conditions=SaveConditions(time=Time(seconds=1)),
            label="DRV"
        ),
        Rest(
            temperature_degc=35, 
            end_conditions=EndConditions(time=Time(hours=1)),
            save_conditions=SaveConditions(time=Time(seconds=1))
        ),
       run=4,
       label="drive_cycle"
    ),
    Cycle(
        Rest(
            temperature_degc=25, 
            end_conditions=EndConditions(
                time=Time(hours=5), 
                temperature=TemperatureEndCondition("<", 26)
            )
            save_conditions=SaveCondition(time=Time(minutes=1))
        )
        run=1
        label="final rest"
   )
)

sim = pybamm.Simulation(model, flow=flow)

brosaplanella, 2023-01-09:
Do we want to keep charge and discharge as separate classes or just do `CC` and `CCCV` and include an argument that is "charge" or "discharge"? I can see pros and cons for both options.

Scottmar93, 2023-01-09:
That's a good point. I guess that `ChargeCC` and `DischargeCC` would be very similar classes and so there would be some repeated code. So probably `CC` and `CCCV` is better. 

And it reads quite nicely: 

What's your thoughts on `loop=1` vs `runs=1` vs `iter=1`? 

```python
from pybamm import ChargeCCCV, DischargeCC, Rest, Run

model = pybamm.lithium_ion.SPM()

flow = pybamm.Flow(
   "flow name",
    Cycle(
        CC("discharge", at="C/2", until="3.2V", save="1s"),
        CCCV("charge", at=("C/2", "3.7V", "25oC"), until=("C/50", "5Ah"), ref="CH1"),
        Rest(at="25oC", until="1h 30m", save="1m"),
        Rest(at="35oC", until="3h", save="1m"),
        loop=1
        ref="init"
    ), 
    Cycle(
        Run(at=("US06.csv", "35oC"), until=("3.2V", "0.1*CH1"), save="1s", ref="DRV"),
        Rest(at="35oC", until="1h", save="1s"), 
        loop=4,
        ref="drive cycle"
    ),
    Cycle(
        Rest(at="25oC", until=("5h", "T<26oC"), save="1m"),
        loop=1
        ref="final rest"
    )
)

cell = pybamm.Cell(model, parmater_values)
cycler = pybamm.Cycler(cell, flow)

```

tinosulzer, 2023-03-09:
# Hackathon: Experiment class

The task for the hackathon is to implement experiment steps as a class, as a complement to the existing string functionality. An experiment created like this should write to the same internal representation as the existing string-based implementation. For example, the experiment
```python
pybamm.Experiment(
    [
        (
            "Discharge at 1C until 2.5 V",
            "Rest for 1 hour",
            "Charge at 5 A until 4.2 V",
            "Hold at 4.2 V until 10 mA",
            "Rest for 1 hour",
        ),
    ]
)
```

converts to 
```python
{'C-rate input [-]': 1.0,
  'dc_data': None,
  'events': {'Voltage input [V]': 2.5, 'type': 'voltage'},
  'period': 60.0,
  'string': 'Discharge at 1C until 2.5 V',
  'tags': None,
  'temperature': None,
  'time': None,
  'type': 'C-rate'},
 {'Current input [A]': 0,
  'dc_data': None,
  'events': None,
  'period': 60.0,
  'string': 'Rest for 1 hour',
  'tags': None,
  'temperature': None,
  'time': 3600.0,
  'type': 'current'},
 {'Current input [A]': -5.0,
  'dc_data': None,
  'events': {'Voltage input [V]': 4.2, 'type': 'voltage'},
  'period': 60.0,
  'string': 'Charge at 5 A until 4.2 V',
  'tags': None,
  'temperature': None,
  'time': None,
  'type': 'current'},
 {'Voltage input [V]': 4.2,
  'dc_data': None,
  'events': {'Current input [A]': 0.01, 'type': 'current'},
  'period': 60.0,
  'string': 'Hold at 4.2 V until 10 mA',
  'tags': None,
  'temperature': None,
  'time': None,
  'type': 'voltage'},
 {'Current input [A]': 0,
  'dc_data': None,
  'events': None,
  'period': 60.0,
  'string': 'Rest for 1 hour',
  'tags': None,
  'temperature': None,
  'time': 3600.0,
  'type': 'current'}]
```

The following tasks should be done by the end of the session. The first one should be done first by the whole group together. The following four could be done in parallel

- [ ] Brainstorm how the new experiment step class should look. The goal here is to converge on a design for a **minimum** viable product, not something that includes every feature that anyone wants. See ideas in the issue above.
- [ ] Implement the experiment step class with conversion to the internal PyBaMM experiment representation.
- [ ] Add tests, make sure tests pass, make sure coverage passes
- [ ] Add documentation to the class and relevant section in the pybamm docs
- [ ] Add an example

Bonus tasks

- [ ] Outline API suggestion for additional features the new experiment step class (or pybamm experiments in general) should have
- [ ] Create formal API spec (in OpenAPI) to describe a generic "battery experiment", which can be used either as PyBaMM's internal representation or for controlling battery cyclers
- [ ] Improve (or suggest improvements for) the way experiments are handled internally by PyBaMM

closed 2023-06-10 by tinosulzer