pybamm-team/PyBaMM #2332: Error while using pybamm.ParameterValues("Chen2020")

Sanjeev1683, 2022-09-30:
I am getting the issue while doing Tutorial number 4, when using the command: parameter_values = pybamm.ParameterValues("Chen2020"). I got the following error message:

---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In [36], line 1
----> 1 parameter_values = pybamm.ParameterValues("Chen2020")

File C:\Program Files\PyBaMM\env\lib\site-packages\pybamm\parameters\parameter_values.py:76, in ParameterValues.__init__(self, values, chemistry)
     73 if values is not None:
     74     # If base_parameters is a filename, load from that filename
     75     if isinstance(values, str):
---> 76         file_path = self.find_parameter(values)
     77         path = os.path.split(file_path)[0]
     78         values = self.read_parameters_csv(file_path)

File C:\Program Files\PyBaMM\env\lib\site-packages\pybamm\parameters\parameter_values.py:756, in ParameterValues.find_parameter(path)
    754     if os.path.isfile(trial_path):
    755         return trial_path
--> 756 raise FileNotFoundError("Could not find parameter {}".format(path))

FileNotFoundError: Could not find parameter Chen2020

tinosulzer, 2022-09-30:
Please check and update pybamm version

krishnakantsingh14, 2022-10-01:
try 
```
pip uninstall pybamm
pip install pybamm -U
```

closed 2022-10-04 by tinosulzer