pybamm-team/PyBaMM #2333: Add single-layer SEI

tinosulzer, 2022-09-30:
Currently the default for SEI is to have an inner and outer layer. This is only for historical reasons because Scott Marquis was investigating this in his [thesis]. 

In practice most people set both layers to be equal or set the inner layer to be zero. It would be good to have the option to only have a single layer SEI. This should probably be the default.

iatzak, 2022-12-04:
I would like to work on this. Can you please assign this issue to me?

tinosulzer, 2022-12-04:
Thanks @iatzak ! Let's work through required changes here before you write any code. If you have ideas on what to do please write out your plan here. If you're unsure or have any questions feel free to ask

iatzak, 2022-12-05:
Great, thanks for the availability to answer questions! I'm attending a [course](https://simulation-software-engineering.github.io/homepage/) where the students are supposed to contribute to an open-source simulation software, and I have chosen PyBaMM. Until the Christmas break I'll be working on a report about PyBAMM's "infrastructure" (CI / documentation / building / contribution / git workflow). After the break, I'll get to this issue, firstly by writing here what I plan to do, as you suggested.

tinosulzer, 2022-12-05:
That's awesome! If you can share the report or make it public, let us know what can be better, and contribute to making it better, that would be really helpful

iatzak, 2022-12-05:
Sure! :) I'll contribute with what I can. I should talk to the professor about sharing the report, but I see no reason to be prohibited.

TomTranter, 2023-01-31:
Hi @iatzak, how's it going? We are running a hackathon on 29th of March in Oxford and this issue is potentially one we could work on together. You'd be welcome to join us if you can make it. Tickets are free but limited availability https://www.eventbrite.co.uk/e/pybamm-hackathon-tickets-517334430207?keep_tld=1
If you have already made some progress on this issue then let us know and we will exclude it from the options to work on during the day. Thanks

iatzak, 2023-02-04:
Hi @TomTranter, thanks for the heads up. I'm working on this later than I expected due to university courses, however I still plan to contribute to this issue. 

@tinosulzer could you describe shortly in which files the changes need to be made?

jeromtom, 2023-02-04:
@iatzak
The changes are to be made in this folder `pybamm\models\submodels\interface\sei`
You will have to create a new python file something like single_layer_sei.py with the parameter `L_inner` set to zero and keep that as default, allowing the user to change `L_outer` to zero if needed. @tinosulzer, please confirm.

Documentation for current [SEI models](https://pybamm.readthedocs.io/en/latest/source/api/models/submodels/interface/sei.html)

jeromtom, 2023-02-21:
Hey @iatzak, I really would like to work on this issue, let me know if you are still busy. Thanks.

iatzak, 2023-02-24:
Hi @jeromtom, you can go ahead, since I made my course-related contribution on another issue and I would only start working on this one  in the next week.

GBlanka, 2023-10-13:
Hi @tinosulzer, I would like to work on this issue, could you please assign it to me?

PR #4470 (2024-11-21): make SEI single layer only
# Description

Remove the double-layer SEI and make it single layer only
This PR falls in the category of "breaking changes to the parameter names" so let's discuss how to handle this in the developer meeting - draft only for now

Fixes #2333
Replaces #3920

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which [... cut]
closed 2024-11-21 by kratman