pybamm-team/PyBaMM #2334: Reformat parameters to a single python file (+ csv files if necessary for interpolation?)

tinosulzer, 2022-09-30:
We have been talking about moving back away from the split-by-type format (negative electrode/separator/positive electrode etc) for parameters, and instead displaying each parameter set in a single python file. See below for an example

Input is welcome on whether this is too much of a breaking change. We would still support the old csv file format, but using python files would be recommended instead. 

Pros:
- It makes it much clearer how to add your own parameter set, either in a separate python file or the same one
- All parameters are now defined in one place for a particular parameter set

Cons:
- It's a change
- Can no longer mix and match parameters (but we're realizing this was probably a bad idea all along)

Example file:
```python3
def graphite_LGM50_ocp_Chen2020(sto):
    """
    LG M50 Graphite open-circuit potential as a function of stochiometry, fit taken
    from [1].

    References
    ----------
    .. [1] Chang-Hui Chen, Ferran Brosa Planella, Kieran O’Regan, Dominika Gastol, W.
    Dhammika Widanage, and Emma Kendrick. "Development of Experimental Techniques for
    Parameterization of Multi-scale Lithium-ion Battery Models." Journal of the
    Electrochemical Society 167 (2020): 080534.

    Parameters
    ----------
    sto: :class:`pybamm.Symbol`
        Electrode stochiometry

    Returns
    -------
    :class:`pybamm.Symbol`
        Open circuit potential
    """

    u_eq = (
        1.9793 * exp(-39.3631 * sto)
        + 0.2482
        - 0.0909 * tanh(29.8538 * (sto - 0.1234))
        - 0.04478 * tanh(14.9159 * (sto - 0.2769))
        - 0.0205 * tanh(30.4444 * (sto - 0.6103))
    )

    return u_eq

def graphite_LGM50_electrolyte_exchange_current_density_Chen2020(
    c_e, c_s_surf, c_s_max, T
):
    """
    Exchange-current density for Butler-Volmer reactions between graphite and LiPF6 in
    EC:DMC.

    References
    ----------
    .. [1] Chang-Hui Chen, Ferran Brosa Planella, Kieran O’Regan, Dominika Gastol, W.
    Dhammika Widanage, and Emma Kendrick. "Development of Experimental Techniques for
    Parameterization of Multi-scale Lithium-ion Battery Models." Journal of the
    Electrochemical Society 167 (2020): 080534.

    Parameters
    ----------
    c_e : :class:`pybamm.Symbol`
        Electrolyte concentration [mol.m-3]
    c_s_surf : :class:`pybamm.Symbol`
        Particle concentration [mol.m-3]
    c_s_max : :class:`pybamm.Symbol`
        Maximum particle concentration [mol.m-3]
    T : :class:`pybamm.Symbol`
[... 225 lines cut ...]
        "Positive electrode porosity": 0.335,
        "Positive electrode reaction-driven LAM factor [m3.mol-1]": 0.0,
        "Positive electrode specific heat capacity [J.kg-1.K-1]": 700.0,
        "Positive electrode thermal conductivity [W.m-1.K-1]": 2.1,
        "Positive electrode thickness [m]": 7.56e-05,
        "Positive particle radius [m]": 5.22e-06,
        "Ratio of lithium moles to SEI moles": 2.0,
        "Reference temperature [K]": 298.15,
        "SEI growth activation energy [J.mol-1]": 0.0,
        "SEI kinetic rate constant [m.s-1]": 1e-12,
        "SEI open-circuit potential [V]": 0.4,
        "SEI reaction exchange current density [A.m-2]": 1.5e-07,
        "SEI resistivity [Ohm.m]": 200000.0,
        "Separator Bruggeman coefficient (electrolyte)": 1.5,
        "Separator density [kg.m-3]": 397.0,
        "Separator porosity": 0.47,
        "Separator specific heat capacity [J.kg-1.K-1]": 700.0,
        "Separator thermal conductivity [W.m-1.K-1]": 0.16,
        "Separator thickness [m]": 1.2e-05,
        "Total heat transfer coefficient [W.m-2.K-1]": 10.0,
        "Typical current [A]": 5.0,
        "Typical electrolyte concentration [mol.m-3]": 1000.0,
        "Upper voltage cut-off [V]": 4.2,
    }
```

brosaplanella, 2022-10-03:
I think it is a good idea. We could still keep the "separation" by organising the dictionary similar to files, with comments as subtitles (e.g. electrolyte, postivie electrode...)

tinosulzer, 2022-10-07:
> for interpolation I would define the list of numbers directly in python file, at least for the parameter sets in pybamm itself.

As a block comment then read in by numpy/pandas? Or directly as a numpy array? What about 2D data, currently defined as a json? https://github.com/pybamm-team/PyBaMM/blob/develop/tests/unit/test_parameters/lico2_diffusivity_Dualfoil1998_2D.json

martinjrobins, 2022-10-08:
directly as a 1D or 2D numpy array perhaps?

PR #2342 (2022-10-11): Issue 2334 reformat parameters
# Description

Change parameters to python files (+ some csv files). Breaking change.

Fixes #2334 
Fixes #2077 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [ ] Bug fix (non-breaking change which fixes an issue)

# [... cut]
closed 2022-10-11 by tinosulzer