pybamm-team/PyBaMM #2338: SEI transfer coefficient

brosaplanella, 2022-10-04:
The SEI transfer coefficient for `reaction limited` and `EC reaction limited` have the transfer coefficient hardcoded to 0.5. This value is a modelling assumption (see [Safari et al. 2009](https://iopscience.iop.org/article/10.1149/1.3043429/pdf)), so I think we should allow to define it in the parameter values so it can be changed.

tinosulzer, 2022-10-04:
Sure. Let's handle it the same way as symmetric/asymmetric BV. Add new SEI options "reaction limited (asymmetric)" and "EC reaction limited (asymmetric)" which require the transfer coefficient to be defined.

brosaplanella, 2022-11-03:
Shall I leave the option `"reaction limited"` (which defaults to the symmetric) with a deprecation warning? Or just introduce a breaking change?

tinosulzer, 2022-11-03:
Definitely have a deprecation warning if it changes. I'm not sure it needs to be changed though?

PR #2425 (2022-11-04): Asymmetric SEI reaction
# Description

Allow for asymmetric SEI reactions, in the cases of `"reaction limited"` and `"ec reaction limited"`. The option is set to `"reaction limited (symmetric)"` and `"reaction limited (asymmetric)"` (same for EC).

Fixes #2338

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [x] New feature (non-breaking change which adds functionality)
- [ [... cut]
closed 2022-11-04 by brosaplanella