pybamm-team/PyBaMM #2341: Remove degradation/thermal parameters from parameter sets where they are not actually fitted

tinosulzer, 2022-10-04:
And add utility functions to make it easy to augment a parameter set with *example* parameters, but make it clear that they are not fitted for that parameter set.

This might break some codes, but we need to make it clear that the degradation parameters are not fitted to every chemistry.

tinosulzer, 2022-10-05:
@DrSOKane in the OKane2020 and OKane2022 parameter sets, which degradation parameters (SEI/plating) are actually referenced in the papers, and which ones are added as examples only (if any)?

DrSOKane, 2022-10-05:
> @DrSOKane in the OKane2020 and OKane2022 parameter sets, which degradation parameters (SEI/plating) are actually referenced in the papers, and which ones are added as examples only (if any)?

I can think of the following:
1) The volume change parameters for the negative and positive electrodes. The mechanics code returns an error if this is not provided, but it doesn't affect the degradation in any way. The expressions are taken from Ai et al. 2020.
2) The cell thermal expansion coefficient, also taken from Ai et al. 2020.
3) The negative electrode OCV function, which OKane2022 is not set up to use by default but I put it there anyway. I've since come up with a better one using additional unpublished data from Kieran.