pybamm-team/PyBaMM #2346: [Bug]: Installing `PyBaMM` on `M1` chip fails (both `pip` and install from source)

Saransh-cpp, 2022-10-05:
### PyBaMM Version

all

### Python Version

3.9

### Describe the bug

`PyBaMM` cannot be installed on an `M1` mac from `pip` as well as from source. I think `casadi` [does not support](https://github.com/casadi/casadi/issues/2829) `M1` chips?

I was able to set everything up by installing `casadi` from source, and removing `casadi` as a dependency when installing `PyBaMM` from source. Should this be documented in the installation page?

Also, should we be specifically building `intel` wheels till `casadi` supports `M1` wheels? This way users will get a "No matching distribution found for pybamm" error when installing through `pip` on `M1`, rather than than the installation failing midway.
```
brew install make swig
git clone https://github.com/casadi/casadi.git casadi
cd casadi
git checkout 3.5.5
mkdir build
cd build
cmake -DWITH_PYTHON=ON \
-DWITH_PYTHON3=ON \
-DPYTHON_INCLUDE_DIR=$(python3 -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") \
-DPYTHON_LIBRARY=$(python3 -c "import distutils.sysconfig as sysconfig; print(sysconfig.get_config_var('LIBDIR'))") \
..
make
sudo make install
```

### Steps to Reproduce

```
pip3 install pybamm
```

```
pip3 install -e .
```

### Relevant log output

```shell
ERROR: Could not find a version that satisfies the requirement casadi>=3.5.0 (from pybamm) (from versions: none)
ERROR: No matching distribution found for casadi>=3.5.0
```

tinosulzer, 2022-10-05:
I'm able to install pybamm on M1 from source, but maybe it's using Rosetta?

Saransh-cpp, 2022-10-05:
I think so. #2072 looks related (`casadi` not found), the only difference is that I did not try building up the `IDAKLUSolver`.

abillscmu, 2022-11-06:
You can force it to use rosetta if you're using conda by setting the environment variable `CONDA_SUBDIR` to `osx-x64`.

savvn001, 2022-11-24:
I managed to get it to install and sort of work.
Using an M1 MBA (2020):

- Download and install Python 3.8.10 **universal** installer (maybe 3.9 works too, I haven't tried)
- Use the interpreter `/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8-intel64` to create a venv and run the pip install pybamm

I think using the intel64 binary, which I guess automatically runs under rosetta, seems to work? When I run a script after doing `import pybamm` it does take a ~4 secs to start running. Not sure if this is how it usually is with this package.

ndrewwang, 2022-12-16:
@savvn001's method on Python 3.8.10 works for me - it did not work with 3.9.

Specifically:
0. From within a rosetta terminal:
1. Installed 3.8.10 from https://www.python.org/downloads/release/python-3810/
2. Created a virtual environment based on the intel64 interpreter: `/Library/Frameworks/Python.framework/Versions/3.8/bin/python3.8-intel64 -m venv venv38intel`
3. For usage in jupyter lab, added the kernel `ipython kernel install --user --name=venv39intel`
4. Activated the venv `source venv38intel/bin/activate`

abillscmu, 2022-12-16:
conda-forge should also work (no rosetta required)

rtimms, 2023-02-07:
FYI the following worked for me on Python 3.9.13 + M2 Macbook Air + MacOS 13.2

1. Installed Python 3.9.13 universal installer from [here](https://www.python.org/ftp/python/3.9.13/python-3.9.13-macos11.pkg)
2. Created a virtual env using the intel64 interpreter `Library/Frameworks/Python.framework/Versions/3.9/bin/python3-intel64 -m venv env`
3. Activate and install as normal

CLOUDUH, 2023-03-01:
MacBook Pro 14 2021 (M1 Pro chip) / MacOS 12.6 /

abillscmu, 2023-03-02:
Out of curiosity has anyone checked if there's a performance difference between Conda/build-from-source and the rosetta options?

agriyakhetarpal, 2023-04-06:
Casadi 3.6.0 brings support for Apple Silicon: https://github.com/casadi/casadi/releases/tag/3.6.0

closed 2023-04-07 by Saransh-cpp