pybamm-team/PyBaMM #2351: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.[Bug]:

Ma2922, 2022-10-07:
### PyBaMM Version

22.5

### Python Version

3.7

### Describe the bug

 I run the model with the same parameters,When the charging current  are 3.3A, 10A and 30A, I get no error, only A warning, but when the charging current is 50A, I get an error.

### Steps to Reproduce

### code

t_eval=np.arange(0,500000,1)
sim_char_03C=pybamm.Simulation(model,parameter_values=params,experiment=char_03C)
sol_char_03C=sim_char_03C.solve(t_eval,initial_soc=0)
sim_char_1C=pybamm.Simulation(model,parameter_values=params,experiment=char_1C)
sol_char_1C=sim_char_1C.solve(t_eval,initial_soc=0)
sim_char_3C=pybamm.Simulation(model,parameter_values=params,experiment=char_3C)
sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)
### error

2022-10-07 21:35:54.412 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
2022-10-07 21:36:07.421 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
2022-10-07 21:36:19.509 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-10-07 21:36:31.898 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-10-07 21:36:33.162 - [ERROR] callbacks.on_experiment_error(226): Simulation error: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

SolverError                               Traceback (most recent call last)
Input In [353], in <cell line: 10>()
      8 sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
      9 sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
---> 10 sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:848, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    846 # If none of the cycles worked, raise an error
    847 if cycle_num == 1 and step_num == 1:
--> 848     raise error
    849 # Otherwise, just stop this cycle
    850 break

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:826, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    824 npts = max(int(round(dt / exp_inputs["period"])) + 1, 2)
    825 try:
--> 826     step_solution = solver.step(
    827         current_solution,
    828         model,
    829         dt,
    830         npts=npts,
    831         save=False,
    832         **kwargs,
    833     )
    834 except pybamm.SolverError as error:
    835     if (
    836         "non-positive at initial conditions" in error.message
    837         and "[experiment]" in error.message
    838     ):

File F:\ananconda2\lib\site-packages\pybamm\solvers\base_solver.py:1206, in BaseSolver.step(self, old_solution, model, dt, npts, external_variables, inputs, save)
   1199 pybamm.logger.verbose(
   1200     "Stepping for {:.0f} < t < {:.0f}".format(
   1201         t * model.timescale_eval,
   1202         (t + dt_dimensionless) * model.timescale_eval,
   1203     )
   1204 )
   1205 timer.reset()
-> 1206 solution = self._integrate(model, t_eval, ext_and_inputs)
   1207 solution.solve_time = timer.time()
   1209 # Check if extrapolation occurred

File F:\ananconda2\lib\site-packages\pybamm\solvers\casadi_solver.py:234, in CasadiSolver._integrate(self, model, t_eval, inputs_dict)
    232         t_dim = t * model.timescale_eval
    233         dt_max_dim = dt_max * model.timescale_eval
--> 234         raise pybamm.SolverError(
    235             f"Maximum number of decreased steps occurred at t={t_dim}. "
    236             "Try solving the model up to this time only or reducing "
    237             f"dt_max (currently, dt_max={dt_max_dim}) and/or reducing "
    238             "the size of the time steps or period of the experiment."
    239         )
    240 # Check if the sign of an event changes, if so find an accurate
    241 # termination point and exit
    242 current_step_sol = self._solve_for_event(current_step_sol)

SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

tinosulzer, 2022-10-07:
Could you test with v22.9? There have been a few improvements to the solver recently to make it more stable

Ma2922, 2022-10-08:
it's been tested with v22.9,but still have the same error as above.
the specific code is as follows:

model=pybamm.lithium_ion.DFN(options={"thermal":"lumped"})
file="C:/Users/lenovo/Desktop/parameters.csv"
value_path=pybamm.get_parameters_filepath(file)
params=pybamm.ParameterValues(value_path)

char_03C=pybamm.Experiment([("Charge at 3.3 A until 4.2 V",                      
                               )])
char_1C=pybamm.Experiment([("Charge at 10 A until 4.2 V",                           
                               )])
char_3C=pybamm.Experiment([("Charge at 30 A until 4.2 V",                           
                               )])
char_5C=pybamm.Experiment([("Charge at 50 A until 4.2 V",                          
                               )])

t_eval=np.arange(0,500000,1)
sim_char_03C=pybamm.Simulation(model,parameter_values=params,experiment=char_03C)
sol_char_03C=sim_char_03C.solve(t_eval,initial_soc=0)
sim_char_1C=pybamm.Simulation(model,parameter_values=params,experiment=char_1C)
sol_char_1C=sim_char_1C.solve(t_eval,initial_soc=0)
sim_char_3C=pybamm.Simulation(model,parameter_values=params,experiment=char_3C)
sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

warning and error:

2022-10-08 09:37:49.879 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment 2022-10-08 09:38:03.044 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment The linesearch algorithm failed with too small a step. 2022-10-08 09:38:15.493 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment The linesearch algorithm failed with too small a step. The linesearch algorithm failed with too small a step. The linesearch algorithm failed with too small a step. 2022-10-08 09:38:27.333 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment The linesearch algorithm failed with too small a step. The linesearch algorithm failed with too small a step. The linesearch algorithm failed with too small a step. The linesearch algorithm failed with too small a step. 2022-10-08 09:38:28.718 - [ERROR] callbacks.on_experiment_error(226): Simulation error: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment. 

--------------------------------------------------------------------------- SolverError                               Traceback (most recent call last) Input In [18], in <cell line: 10&gt;()       8 sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)       9 sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C) ---&gt; 10 sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0) File F:\ananconda2\lib\site-packages\pybamm\simulation.py:848, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)     846 # If none of the cycles worked, raise an error     847 if cycle_num == 1 and step_num == 1: --&gt; 848     raise error     849 # Otherwise, just stop this cycle     850 break File F:\ananconda2\lib\site-packages\pybamm\simulation.py:826, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)     824 npts = max(int(round(dt / exp_inputs["period"])) + 1, 2)     825 try: --&gt; 826     step_solution = solver.step(     827         current_solution,     828         model,     829         dt,     830         npts=npts,     831         save=False,     832         **kwargs,     833     )     834 except pybamm.SolverError as error:     835     if (     836         "non-positive at initial conditions" in error.message     837         and "[experiment]" in error.message     838     ): File F:\ananconda2\lib\site-packages\pybamm\solvers\base_solver.py:1206, in BaseSolver.step(self, old_solution, model, dt, npts, external_variables, inputs, save)    1199 pybamm.logger.verbose(    1200     "Stepping for {:.0f} < t < {:.0f}".format(    1201         t * model.timescale_eval,    1202         (t + dt_dimensionless) * model.timescale_eval,    1203     )    1204 )    1205 timer.reset() -&gt; 1206 solution = self._integrate(model, t_eval, ext_and_inputs)    1207 solution.solve_time = timer.time()    1209 # Check if extrapolation occurred File F:\ananconda2\lib\site-packages\pybamm\solvers\casadi_solver.py:234, in CasadiSolver._integrate(self, model, t_eval, inputs_dict)     232         t_dim = t * model.timescale_eval     233         dt_max_dim = dt_max * model.timescale_eval --&gt; 234         raise pybamm.SolverError(     235             f"Maximum number of decreased steps occurred at t={t_dim}. "     236             "Try solving the model up to this time only or reducing "     237             f"dt_max (currently, dt_max={dt_max_dim}) and/or reducing "     238             "the size of the time steps or period of the experiment."     239         )     240 # Check if the sign of an event changes, if so find an accurate     241 # termination point and exit     242 current_step_sol = self._solve_for_event(current_step_sol) SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment. 

抄送: "BIT ***@***.******@***.***&gt;;

Ma2922, 2022-10-08:
It's been testwd with v22.9,but still have the same error as above.

The specific code is as follows:

model=pybamm.lithium_ion.DFN(options={"thermal":"lumped"})
file="C:/Users/lenovo/Desktop/parameters.csv"
value_path=pybamm.get_parameters_filepath(file)
params=pybamm.ParameterValues(value_path)
char_03C=pybamm.Experiment([("Charge at 3.3 A until 4.2 V",
                               )])
char_1C=pybamm.Experiment([("Charge at 10 A until 4.2 V",
                               )])
char_3C=pybamm.Experiment([("Charge at 30 A until 4.2 V",
                               )])
char_5C=pybamm.Experiment([("Charge at 50 A until 4.2 V",
                               )])
t_eval=np.arange(0,500000,1)
sim_char_03C=pybamm.Simulation(model,parameter_values=params,experiment=char_03C)
sol_char_03C=sim_char_03C.solve(t_eval,initial_soc=0)
sim_char_1C=pybamm.Simulation(model,parameter_values=params,experiment=char_1C)
sol_char_1C=sim_char_1C.solve(t_eval,initial_soc=0)
sim_char_3C=pybamm.Simulation(model,parameter_values=params,experiment=char_3C)
sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

Warning:

2022-10-08 09:37:49.879 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
2022-10-08 09:38:03.044 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
2022-10-08 09:38:15.493 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-10-08 09:38:27.333 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-10-08 09:38:28.718 - [ERROR] callbacks.on_experiment_error(226): Simulation error: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

error:

SolverError                               Traceback (most recent call last)
Input In [18], in <cell line: 10>()
      8 sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
      9 sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
---> 10 sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:848, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    846 # If none of the cycles worked, raise an error
    847 if cycle_num == 1 and step_num == 1:
--> 848     raise error
    849 # Otherwise, just stop this cycle
    850 break

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:826, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    824 npts = max(int(round(dt / exp_inputs["period"])) + 1, 2)
    825 try:
--> 826     step_solution = solver.step(
    827         current_solution,
    828         model,
    829         dt,
    830         npts=npts,
    831         save=False,
    832         **kwargs,
    833     )
    834 except pybamm.SolverError as error:
    835     if (
    836         "non-positive at initial conditions" in error.message
    837         and "[experiment]" in error.message
    838     ):

File F:\ananconda2\lib\site-packages\pybamm\solvers\base_solver.py:1206, in BaseSolver.step(self, old_solution, model, dt, npts, external_variables, inputs, save)
   1199 pybamm.logger.verbose(
   1200     "Stepping for {:.0f} < t < {:.0f}".format(
   1201         t * model.timescale_eval,
   1202         (t + dt_dimensionless) * model.timescale_eval,
   1203     )
   1204 )
   1205 timer.reset()
-> 1206 solution = self._integrate(model, t_eval, ext_and_inputs)
   1207 solution.solve_time = timer.time()
   1209 # Check if extrapolation occurred

File F:\ananconda2\lib\site-packages\pybamm\solvers\casadi_solver.py:234, in CasadiSolver._integrate(self, model, t_eval, inputs_dict)
    232         t_dim = t * model.timescale_eval
    233         dt_max_dim = dt_max * model.timescale_eval
--> 234         raise pybamm.SolverError(
    235             f"Maximum number of decreased steps occurred at t={t_dim}. "
    236             "Try solving the model up to this time only or reducing "
    237             f"dt_max (currently, dt_max={dt_max_dim}) and/or reducing "
    238             "the size of the time steps or period of the experiment."
    239         )
    240 # Check if the sign of an event changes, if so find an accurate
    241 # termination point and exit
    242 current_step_sol = self._solve_for_event(current_step_sol)

SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

Ma2922, 2022-10-08:
When I try to charge at 50A  in a very short time,it's still have an error.

code

t_eval=np.arange(0,0.5,0.1)
sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)
 
Warning

2022-10-08 15:16:47.389 - [WARNING] simulation.solve(721): Ignoring t_eval as solution times are specified by the experiment
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
The linesearch algorithm failed with too small a step.
2022-10-08 15:16:48.902 - [ERROR] callbacks.on_experiment_error(226): Simulation error: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

error

SolverError                               Traceback (most recent call last)
Input In [35], in <cell line: 10>()
      3 #sim_char_03C=pybamm.Simulation(model,parameter_values=params,experiment=char_03C)
      4 #sol_char_03C=sim_char_03C.solve(t_eval,initial_soc=0)
      5 #sim_char_1C=pybamm.Simulation(model,parameter_values=params,experiment=char_1C)
      6 #sol_char_1C=sim_char_1C.solve(t_eval,initial_soc=0)
      7 #sim_char_3C=pybamm.Simulation(model,parameter_values=params,experiment=char_3C)
      8 #sol_char_3C=sim_char_3C.solve(t_eval,initial_soc=0)
      9 sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
---> 10 sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:848, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    846 # If none of the cycles worked, raise an error
    847 if cycle_num == 1 and step_num == 1:
--> 848     raise error
    849 # Otherwise, just stop this cycle
    850 break

File F:\ananconda2\lib\site-packages\pybamm\simulation.py:826, in Simulation.solve(self, t_eval, solver, check_model, save_at_cycles, calc_esoh, starting_solution, initial_soc, callbacks, **kwargs)
    824 npts = max(int(round(dt / exp_inputs["period"])) + 1, 2)
    825 try:
--> 826     step_solution = solver.step(
    827         current_solution,
    828         model,
    829         dt,
    830         npts=npts,
    831         save=False,
    832         **kwargs,
    833     )
    834 except pybamm.SolverError as error:
    835     if (
    836         "non-positive at initial conditions" in error.message
    837         and "[experiment]" in error.message
    838     ):

File F:\ananconda2\lib\site-packages\pybamm\solvers\base_solver.py:1206, in BaseSolver.step(self, old_solution, model, dt, npts, external_variables, inputs, save)
   1199 pybamm.logger.verbose(
   1200     "Stepping for {:.0f} < t < {:.0f}".format(
   1201         t * model.timescale_eval,
   1202         (t + dt_dimensionless) * model.timescale_eval,
   1203     )
   1204 )
   1205 timer.reset()
-> 1206 solution = self._integrate(model, t_eval, ext_and_inputs)
   1207 solution.solve_time = timer.time()
   1209 # Check if extrapolation occurred

File F:\ananconda2\lib\site-packages\pybamm\solvers\casadi_solver.py:234, in CasadiSolver._integrate(self, model, t_eval, inputs_dict)
    232         t_dim = t * model.timescale_eval
    233         dt_max_dim = dt_max * model.timescale_eval
--> 234         raise pybamm.SolverError(
    235             f"Maximum number of decreased steps occurred at t={t_dim}. "
    236             "Try solving the model up to this time only or reducing "
    237             f"dt_max (currently, dt_max={dt_max_dim}) and/or reducing "
    238             "the size of the time steps or period of the experiment."
    239         )
    240 # Check if the sign of an event changes, if so find an accurate
    241 # termination point and exit
    242 current_step_sol = self._solve_for_event(current_step_sol)

SolverError: Maximum number of decreased steps occurred at t=0.0. Try solving the model up to this time only or reducing dt_max (currently, dt_max=12.199448838323828) and/or reducing the size of the time steps or period of the experiment.

rtimms, 2022-10-08:
Note that `t_eval` is ignored when you’re using an experiment. You should try reducing the period instead.

Ma2922, 2022-10-08:
Sorry,I don't know what's the mean of reducing the period.Can you talk more specific?

抄送: "BIT ***@***.******@***.***&gt;;

Ma2922, 2022-10-09:
code:
char_5C=pybamm.Experiment([("Charge at 50 A until 4.2 V")])
t_eval=np.arange(0,500000,1)
sim_char_5C=pybamm.Simulation(model,parameter_values=params,experiment=char_5C)
sol_char_5C=sim_char_5C.solve(t_eval,initial_soc=0)

When the model  charge at 5C,my experiment only have one step,I don't know what's the mean of reduce the period.Can you tell me how to operate?

rtimms, 2022-10-10:
When you use an experiment `t_eval` is ignored. Instead the solution is returned at times based on the period. The default is 60s. You can change it for the whole experiment by passing a keyword argument
```python3
experiment = pybamm.Experiment(
    [
        (
            "Discharge at C/5 for 10 hours or until 3.3 V",
            "Rest for 1 hour",
            "Charge at 1 A until 4.1 V",
            "Hold at 4.1 V until 10 mA",
            "Rest for 1 hour",
        ),
    ]
    * 3,
    period="10 s",
)
```
or control it for each step by adding the period in brackets
```python3
experiment = pybamm.Experiment(
    [
        (
            "Discharge at C/5 for 10 hours or until 3.3 V (10 s period)",
            "Rest for 1 hour (120 s period) ",
            "Charge at 1 A until 4.1 V",
            "Hold at 4.1 V until 10 mA",
            "Rest for 1 hour",
        ),
    ]
    * 3,
)
```

You might also find you need to look at some of the extra solver options - the solver is failing at the first step.

Ma2922, 2022-10-23:
Thanks for your detail answer,I will look at the extra solver options.

closed 2023-05-15 by rtimms