pybamm-team/PyBaMM #2368: Overload numpy functions

tinosulzer, 2022-10-14:
There is a way to write `np.exp(symbol)` (and other functions) to return what is currently returned by `pybamm.exp(symbol)`. This would be useful to implement so that functions can be defined using np instead of pybamm

https://towardsdatascience.com/wrapping-numpys-arrays-971e015e14bb

agriyakhetarpal, 2023-02-04:
Maybe I can take this up @tinosulzer

tinosulzer, 2023-02-06:
Great, let me know if you need any help

agriyakhetarpal, 2023-02-14:
I went through the blog in the issue comment and tried to understand it. The tutorial was for regular numpy functions – but since `np.exp` is a universal numpy function, I will have to use the `__array_ufunc__` interface in a similar way as described.

If I get it correctly, do I need to modify the class below to work with `np.exp`? 
https://github.com/pybamm-team/PyBaMM/blob/bf0bdfe682e767171363a92396d8575dfebe0b14/pybamm/expression_tree/functions.py#L352-L365

Saransh-cpp, 2023-02-14:
I've pushed some initial experiments I did on this here - https://github.com/Saransh-cpp/PyBaMM/tree/np-funcs ([diff here](https://github.com/pybamm-team/PyBaMM/compare/develop...Saransh-cpp:PyBaMM:np-funcs?expand=1)). If I remember correctly, everything new was working fine but overloading the `ufunc`s broke some existing code (tests (not the newly added ones) were failing).

Also, for more code to look at: [vector](https://github.com/scikit-hep/vector) overloads -
- `ufunc` here - https://github.com/scikit-hep/vector/blob/120d2ab7da1c0299170ebc80544ec0338823554b/src/vector/backends/numpy.py#L710
- and `functions` here - https://github.com/scikit-hep/vector/blob/120d2ab7da1c0299170ebc80544ec0338823554b/src/vector/backends/numpy.py#L925.

jaskiratsingh2000, 2023-02-20:
Is this still open?

tinosulzer, 2023-02-20:
Yes, but someone else is assigned

agriyakhetarpal, 2023-02-25:
I tried some initial experiments as well, I think a recipe like this should work, if I am on the right path? The ufuncs include a variety of functions so testing them for a few was my initial try

```python
# overload universal numpy functions for Symbol
    def __array_ufunc__(self, ufunc, method, *inputs, **kwargs):
        # This method is called by numpy when a ufunc is called on a Symbol
        # It is called instead of the ufunc's __call__ method
        # We use this to apply the ufunc to the Symbol's value, rather than the
        # Symbol itself
        # This allows us to use numpy functions on Symbols, e.g. np.sin(pybamm.Scalar(1))
        if method == "__call__":
            # Get the value of the Symbol
            value = self.evaluate()
            # Apply the ufunc to the value
            result = ufunc(value, *inputs[1:], **kwargs)
            # If the result is a scalar, return a Scalar
            if np.isscalar(result):
                return pybamm.Scalar(result)
            # Otherwise, return a Vector
            else:
                return pybamm.Vector(result)
        else:
            raise NotImplementedError(
                f"Symbol does not support ufunc method {method}"
            )
```

Otherwise we may construct 
It would help if I could

agriyakhetarpal, 2023-02-25:
I tried some initial experiments as well – I think a recipe like this should work, if I am on the right path? The ufuncs include a variety of functions so testing a few of them was my initial try

```python
# overload universal numpy functions for Symbol
    def __array_ufunc__(self, ufunc, method, *inputs, **kwargs):
        # This method is called by numpy when a ufunc is called on a Symbol
        # It is called instead of the ufunc's __call__ method
        # We use this to apply the ufunc to the Symbol's value, rather than the
        # Symbol itself
        # This allows us to use numpy functions on Symbols, e.g. np.sin(pybamm.Scalar(1))
        if method == "__call__":
            # Get the value of the Symbol
            value = self.evaluate()
            # Apply the ufunc to the value
            result = ufunc(value, *inputs[1:], **kwargs)
            # If the result is a scalar, return a Scalar
            if np.isscalar(result):
                return pybamm.Scalar(result)
            # Otherwise, return a Vector
            else:
                return pybamm.Vector(result)
        else:
            raise NotImplementedError(
                f"Symbol does not support ufunc method {method}"
            )
```

Otherwise, perhaps we may construct inputs –> arrays –> output after the ufuncs have been applied
```python
# overload universal numpy functions for class:`Symbol`
    def __array_ufunc__(self, ufunc, method, *inputs, **kwargs):
        """Overload universal numpy functions for class:`Symbol`."""
        # Convert inputs to pybamm objects
        inputs = tuple(pybamm.Scalar(x) if isinstance(x, Number) else x for x in inputs)
        # Convert outputs to numpy arrays
        outputs = kwargs.pop("out", None)
        if outputs:
            kwargs["out"] = tuple(
                o._data if isinstance(o, pybamm.Symbol) else o for o in outputs
            )
        # Evaluate ufunc
        result = getattr(ufunc, method)(*inputs, **kwargs)
        # Convert result to pybamm object
        if isinstance(result, tuple):
            result = tuple(
                pybamm.Scalar(x) if isinstance(x, Number) else x for x in result
            )
        else:
            result = pybamm.Scalar(result) if isinstance(result, Number) else result
        return result
```

I still don't think I understand this fully

tinosulzer, 2023-05-16:
I haven't tested this extensively, but adding this to the `Symbol` class seems to work:
```python
def __array_ufunc__(self, ufunc, method, *inputs, **kwargs):
        return getattr(pybamm, ufunc.__name__)(*inputs, **kwargs)
```

tinosulzer, 2023-07-30:
@ejfdickinson please take a look at the branch from the associated PR and let us know if it fixes the specific use case you have in mind

PR #3205 (2023-08-02): #2368 overload numpy functions
# Description

- Numpy functions can now be called on pybamm symbolic objects. This will return a pybamm symbolic object applying the same function (e.g. `np.exp(x)` returns `pybamm.Exp(x)`). Parameter sets in the library have been changed to use numpy functions, but they do not need to be (the old style of calling pybamm functions will still work)
- Binary operators applied to a pybamm object and numpy array return a pybamm binary operator, where previously it raised an error. The numpy array gets converted to a pybamm Vector. This only works with 1D arrays (it  [... cut]
closed 2023-08-02 by tinosulzer