pybamm-team/PyBaMM #2388: PyBaMM develop does not work on Google Colab (Colab runs on Python 3.7)

Saransh-cpp, 2022-10-20:
Recently https://github.com/pybamm-team/PyBaMM/pull/2379 removed Python `3.7`'s support from `PyBaMM`. Removal of Python `3.7` broke all the `Colab` examples, as Google Colab runs on Python `3.7.15` by default.

Installing the develop version of `PyBaMM` -
```python
%pip install git+https://github.com/pybamm-team/PyBaMM@develop -q  # install the develop branch of PyBaMM

import pybamm
```
errors out -
```python
ERROR: Package 'pybamm' requires a different Python: 3.7.15 not in '<3.10,>=3.8'
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
[<ipython-input-1-7946832af629>](https://8mcarsee425-496ff2e9c6d22116-0-colab.googleusercontent.com/outputframe.html?vrz=colab-20221018-060106-RC00_481858649#) in <module>
      1 get_ipython().run_line_magic('pip', 'install git+https://github.com/pybamm-team/PyBaMM@develop -q  # install the develop branch of PyBaMM')
      2 import os
----> 3 import pybamm
      4 import logging
      5 import requests

ModuleNotFoundError: No module named 'pybamm'

---------------------------------------------------------------------------
NOTE: If your import is failing due to a missing package, you can
manually install dependencies using either !pip or !apt.

To view examples of installing some common dependencies, click the
"Open Examples" button below.
---------------------------------------------------------------------------
```

There is an open issue regarding this in https://github.com/googlecolab/colabtools/ - https://github.com/googlecolab/colabtools/issues/2165.

There are some hacks that make other Python versions run on `Colab`, but none of them is a complete solution.

I am not sure what should be done here. Newcomers might run into this problem, but this is something that we cannot fix.

brosaplanella, 2022-10-20:
Checking the issue in Colab, it seems they are not too pressed for that so we might need to explore some other options if they don't fix it quickly (or pin the PyBaMM version in the getting started examples at least so they still work).

tinosulzer, 2022-10-20:
Very strange that they haven't updated it yet considering numpy has dropped python 3.7 support. We could consider switching to an alternative such as Deepnote or something else

Saransh-cpp, 2022-10-21:
I remember replacing Binder's badge and textual information with Google Colab's before GSoC 😅. I don't recall why PyBaMM shifted to Google Colab, but maybe we can move back?

Also, this might be fixed very soon, given that conda-forge is also removing Python 3.7 support. So maybe we can wait a couple of weeks?

tinosulzer, 2022-10-21:
Yeah, let's see if the conda-forge move triggers a change from Google Colab. We switched to Colab from Binder because it was a lot easier to get running

brosaplanella, 2022-12-05:
It seems it [has been fixed](https://github.com/googlecolab/colabtools/issues/3246) but there is now a new error
```
---------------------------------------------------------------------------
PluginNotFound                            Traceback (most recent call last)
[<ipython-input-2-4478c83dbc15>](https://localhost:8080/#) in <module>
----> 1 import pybamm

6 frames
[/usr/local/lib/python3.8/dist-packages/pybtex/plugin/__init__.py](https://localhost:8080/#) in _load_entry_point(group, name, use_aliases)
     79         for entry_point in pkg_resources.iter_entry_points(search_group, name):
     80             return entry_point.load()
---> 81     raise PluginNotFound(group, name)
     82 
     83 

PluginNotFound: plugin pybtex.database.input.bibtex not found
```

I have opened issue #2524

Saransh-cpp, 2022-12-05:
I'll close this because we have a new error now haha

closed 2022-12-05 by Saransh-cpp