pybamm-team/PyBaMM #2394: 3D Processed Variables

rtimms, 2022-10-21:
Allow processing of 3D variables. E.g. to access concentration in a 1x1x1D model:

```python3
import pybamm

model = pybamm.lithium_ion.DFN(options={"current collector": "potential pair", "dimensionality": 1})
sim = pybamm.Simulation(model)
sol = sim.solve([0, 100])
sol["Negative particle concentration [mol.m-3]"].entries
```

This currently gives `NotImplementedError: Shape not recognized for y[0:4000] * 24983.2619938437 (note processing of 3D variables is not yet implemented)`

rtimms, 2022-10-21:
See #1549

rtimms, 2022-10-21:
Workaround to evaluate 3D variables at a given time and state

```python3
# get expression for the variable of interest
c_n = sim.built_model.variables["Negative particle concentration [mol.m-3]"]

# get time and states
t = sol.t
y = sol.y

# evaluate at a given time and state - the expression takes (time, state, inputs)
c_n.evaluate(t[90], y[:, 90], [])
```

tinosulzer, 2022-10-21:
#2366 might make this easier (especially the interpolation in 3D)

rtimms, 2023-05-02:
should be easier now #2907 has been merged

kratman, 2025-04-02:
@rtimms Closing this the PR did not close it because the `Fixes` tag only takes 1 reference

closed 2025-04-02 by kratman