pybamm-team/PyBaMM #2410: Control automatic function parameter interpolation method

Scottmar93, 2022-10-28:
### Description

Allow users to specify which interpolation method they would like to be applied to function parameters. 

### Motivation

When using three-dimensional lookup tables I have found better model build times when moving from "cubic" to "linear" interpolation (in particular convert_to_casadi is slow). At the moment I can only control this by modifying the process symbol method of parameter values. 

The speed up might be due to differences in calculating the Jacobian (or perhaps not calculating it if the Jacobian at all is not defined everywhere for linear interpolation) but I am not sure of the exact reason. 

### Possible Implementation

I can think of a couple of ways of doing this but I'm not sure what is preferable for everyone. Both should be non-breaking. 

**1. Pass optional dictionary to parameter_values class** 
This would just involve adding an optional argument to the parameter_values class and then adding the input as an attribute of the class. The user would just pass an options dictionary in the following way: 
```python
options = {"interpolation methods": {"name of function parameter to target": "linear"} 
```
if not specified then "cubic" interpolation will be used so that this change is nonbreaking. 

**2. Allow users to add a third entry to the tuple within the parameter values dictionary** 
i.e. 
```python
options = {"interpolation method": "linear"}
parameter_values_dictionary = {
   "negative electrode open circuit voltage [V]": ("negative electrode open circuit voltage [V]", ( (x_data, ), y_data), options), 
   "positive electrode open circuit voltage [V]": ("positive electrode open circuit voltage [V]", ( (x_data, ), y_data)), 
}
```
This could either be a dictionary like I suggest here (so that additional control can be easily added later) or it could just be "linear" or "cubic". This third argument should be optional with "cubic" used by default so that we don't break all previous code. 

### Additional context

I am happy to implement once there is agreement on how to do so. 

Could perhaps be done in conjunction with modifications to function parameter in #2403

rtimms, 2022-10-28:
You can also create the `Interpolant` first to give more control rather than passing data directly, but I think a solution like you suggest would be useful.

tinosulzer, 2022-10-28:
Passing in the interpolant directly might be tricky because you don't know what the children objects are from outside the model classes. However, you can create a function that returns the interpolant object. I would favor this route (examples demonstrating how to do this in 1D, 2D, 3D) rather than functionality that lets you pass options, since it's fairly easy to do yourself if there is an example

It's interesting that you found linear to be better than cubic. I thought cubic worked better because of needing to differentiate? Ideally, the default would be linear.

rtimms, 2022-10-28:
Ah yeah good point. When I changed the default interpolant to linear I left the function parameters as cubic as I thought differentiating would lead to issues. I agree having linear as the default everywhere would be best.

rtimms, 2022-11-01:
Would be interesting to do a bit more detailed profiling to see where the bottleneck is

PR #2510 (2022-12-04): explicitly create interpolants from functions
# Description

In the standard parameter sets, create the interpolants from functions explicitly (instead of relying on the `ParameterValues` class to do it). This allows finer control of hyperparameters (e.g. interpolator).

Using "cubic" interpolator for OCV gives much faster performance than linear interpolator (e.g. 1s -> 0.5s solve time for Ai2020 on my laptop)

Fixes #2410 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (in [... cut]
closed 2022-12-04 by tinosulzer