pybamm-team/PyBaMM #2418: Dimensional models

tinosulzer, 2022-11-01:
Some basic profiling reveals that implementing the models in a dimensional way provides no slow-down compared to dimensionless. Since dimensionless models are a big barrier to entry, we should reformat the models to be entirely dimensional. Further benchmarking and testing will be needed.

Note: scaling variables is still a good idea for the condition number of the solver, but this can be done directly in the solver

tinosulzer, 2022-11-01:
We currently have a lot of duplicate variables (dimensionless/dimensional).
Q: Should we just get rid of the dimensionless ones?

tinosulzer, 2022-11-01:
Would be good to revive #406 when implementing this to add an additional layer of testing

brosaplanella, 2022-11-02:
> We currently have a lot of duplicate variables (dimensionless/dimensional). Q: Should we just get rid of the dimensionless ones?

I think so, I would just keep those that have actually some sort of physical meaning beyond being the dimensionless form of something (e.g. dimensionless particle concentrations / stoichiometry).

PR #2724 (2023-02-23): Divided by 3600 to replace t0_cr
# Description

The old non-dimensionalized PyBaMM had a constant t0_cr used for the crack propagation submodel. This was deleted, so I divided dl_cr/dt by 3600 (standard value for t0_cr) to compensate and retreive the same results as existing PyBaMM.

Fixes #2418 

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking [... cut]
PR #2419 (2023-03-01): Issue 2418 dimensional
# Description

Makes all models dimensional. Major changes:
- Model equations now specified in dimensional form. To help with this, there are a `scale` and `reference` attributes for variables, which can be set to make sure that variables are order 1. Algebraic equations also have to be multiplied by an appropriate number (e.g. `L_x**2`) to make them order 1 and hence pass the tolerance check.
- Removed dimensionless parameters
- Removed most dimensionless variables
- Removed some parameters that were just for non-dimensionalization (typical current, typical electrolyte  [... cut]
closed 2023-03-01 by tinosulzer