pybamm-team/PyBaMM #2422: [Bug]:

Mrzhang-hub, 2022-11-02:
### PyBaMM Version

22.10

### Python Version

3.8

### Describe the bug

```
2022-11-02 21:40:12.211 - [INFO] base_model._build_model(572): Start building Doyle-Fuller-Newman model
2022-11-02 21:40:12.281 - [INFO] base_battery_model.build_model(1019): Finish building Doyle-Fuller-Newman model
2022-11-02 21:40:12.282 - [INFO] parameter_values.process_model(371): Start setting parameters for Doyle-Fuller-Newman model
2022-11-02 21:40:12.389 - [INFO] parameter_values.process_model(474): Finish setting parameters for Doyle-Fuller-Newman model
2022-11-02 21:40:12.392 - [INFO] discretisation.process_model(149): Start discretising Doyle-Fuller-Newman model
2022-11-02 21:40:12.401 - [INFO] discretisation.remove_independent_variables_from_rhs(1199): removing variable Discharge capacity [A.h] from rhs
2022-11-02 21:40:12.859 - [INFO] discretisation.process_model(271): Finish discretising Doyle-Fuller-Newman model
2022-11-02 21:40:12.860 - [INFO] base_solver.solve(703): Start solving Doyle-Fuller-Newman model with CasADi solver with 'safe' mode
2022-11-02 21:40:12.866 - [INFO] base_solver.set_up(110): Start solver set-up
2022-11-02 21:40:13.107 - [INFO] base_solver.set_up(235): Finish solver set-up
CasADi -  WARNING(".../casadi/core/plugin_interface.hpp:322: Assertion "handle!=nullptr" failed:
PluginInterface::load_plugin: Cannot load shared library 'libcasadi_rootfinder_newton.dll': 
   (
    Searched directories: 1. casadipath from GlobalOptions
                          2. CASADIPATH env var
                          3. PATH env var (Windows)
                          4. LD_LIBRARY_PATH env var (Linux)
                          5. DYLD_LIBRARY_PATH env var (osx)
    A library may be 'not found' even if the file exists:
          * library is not compatible (different compiler/bitness)
          * the dependencies are not found
   )
  Tried '...\venv\lib\site-packages\casadi' :
    Error code (WIN32): 126
  Tried '' :
    Error code (WIN32): 126
  Tried '.' :
    Error code (WIN32): 126") [.../casadi/core/plugin_interface.hpp:171]
```

### Steps to Reproduce

```

import pybamm
import numpy as np

pybamm.set_logging_level("INFO")

# load model
model = pybamm.lithium_ion.DFN()
# create geometry
geometry = model.default_geometry

# load parameter values and process model and geometry
param = model.default_parameter_values
param.process_geometry(geometry)
param.process_model(model)

# set mesh
var = pybamm.standard_spatial_vars
var_pts = {var.x_n: 30, var.x_s: 30, var.x_p: 30, var.r_n: 10, var.r_p: 10}
mesh = pybamm.Mesh(geometry, model.default_submesh_types, var_pts)

# discretise model
disc = pybamm.Discretisation(mesh, model.default_spatial_methods)
disc.process_model(model)

# solve model
t_eval = np.linspace(0, 3600, 100)
solver = pybamm.CasadiSolver(mode="safe", atol=1e-6, rtol=1e-3)
solution = solver.solve(model, t_eval)

# plot
plot = pybamm.QuickPlot(
    solution,
    [
        "Current [A]",
        "Terminal voltage [V]",
    ],
    time_unit="seconds",
    spatial_unit="um",
)
plot.dynamic_plot()
```

tinosulzer, 2022-11-02:
Looks like a casadi issue. Does this work?
```
from casadi import *

# Symbols/expressions
x = MX.sym('x')
y = MX.sym('y')
z = MX.sym('z')
f = x**2+100*z**2
g = z+(1-x)**2-y

nlp = {}                 # NLP declaration
nlp['x']= vertcat(x,y,z) # decision vars
nlp['f'] = f             # objective
nlp['g'] = g             # constraints

# Create solver instance
F = nlpsol('F','ipopt',nlp);

# Solve the problem using a guess
F(x0=[2.5,3.0,0.75],ubg=0,lbg=0)
```

Mrzhang-hub, 2022-11-05:
> Looks like a casadi issue. Does this work?
>
> ```
> from casadi import *
>
>
> # Symbols/expressions
> x = MX.sym('x')
> y = MX.sym('y')
> z = MX.sym('z')
> f = x**2+100*z**2
> g = z+(1-x)**2-y
>
> nlp = [...]

Still have an error. 
```

RuntimeError                              Traceback (most recent call last)
Cell In [2], line 17
     14 nlp['g'] = g             # constraints
     16 # Create solver instance
---> 17 F = nlpsol('F','ipopt',nlp);
     19 # Solve the problem using a guess
     20 F(x0=[2.5,3.0,0.75],ubg=0,lbg=0) 
File e:\venv\lib\site-packages\casadi\casadi.py:16847, in nlpsol(*args)
  15857 def nlpsol(*args):
  15858     """
  15859 
  15860 
   (...)
  16845 
  16846     """
> 16847     return _casadi.nlpsol(*args)

RuntimeError: .../casadi/core/plugin_interface.hpp:417: Plugin 'ipopt' is not found.
```

tinosulzer, 2022-11-20:
You should open an issue with casadi to figure out how to fix this

closed 2022-11-20 by tinosulzer