pybamm-team/PyBaMM #2427: Make private functions and classes more explicit

tinosulzer, 2022-11-03:
Make it clearer which methods should not be accessed by users, by adding an underscore at the start of their name

Saransh-cpp, 2024-03-05:
> We should also look into adding \_\_all\_\_ to pybamm modules

Copied from the PR linked above

agriyakhetarpal, 2024-03-05:
There was an attempt by @arjxn-py on adding `__all__` with the `mkinit` CLI-based package to speed up the PyBaMM import, but there were problems with that approach (#3732) because it would be messy to add it to several files at once and generate it everytime. Here is an older thread I found, probably not relevant anymore: #2436 

It would make sense to start with smaller modules like `util.py`, `doc_utils.py`, `install_odes.py` (the latter should just have one method of usage – through the entry point). Also, `version.py` should be a private module too (`pybamm._version.__version__` instead of `pybamm.version.__version__`), and it could be altered in favour of using `importlib.metadata.version()` to fetch the version from declarative metadata – in 2024 we should not need to import a package to obtain its version

arjxn-py, 2024-03-07:
It should be possible to add `__all__` only as well with the help of `mkinit`, how'd we want to have it though - 
- have `__all__` added to all the submodules? (**Not Recommended** - as it would again be messy with several file changes)
- have a recursive `__all__` added to the root `__init__.py` (**Recommended** - See [this commit](https://github.com/arjxn-py/PyBaMM/commit/175788523975a8fbd0200a0e0bc2b14e721075d7))
- have a non-recursive `__all__` added to the root `__init__.py` (See [this commit](https://github.com/arjxn-py/PyBaMM/commit/d71a8a348c0b946e80f8855be18fb18bd8d59600), Makes lesser sense as it only includes direct attributes of pybamm)

agriyakhetarpal, 2024-03-07:
The second one makes the most sense to me, it should decide the namespace when we import PyBaMM.

The third one would need a bit more refining because we have things like `update_LD_LIBRARY_PATH` which should not be exposed

arjxn-py, 2024-03-08:
> The third one would need a bit more refining because we have things like `update_LD_LIBRARY_PATH` which should not be exposed

We also have these attributes in the second too, as ultimately: 
- First is the superset of Second & Third & 
- Second is the superset of Third

Currently Protected modules are exposed, but their attributes are not & Private modules and their attributes are not exposed.

I shall open a PR going forward with the second at the moment but to refine this i'd welcome reviews to omit attributes based upon different preferences.