pybamm-team/PyBaMM #2436: `import pybamm` could be faster

tinosulzer, 2022-11-08:
Doing some basic profiling, a lot (25%) of time is spent initializing the entry points in `parameter_sets.py`. Can this be made faster @awadell1 ? Or only initialized if/when it's first called?

awadell1, 2022-11-08:
Maybe. How much time is spent in `__init__`? And how much on the `.load()` call?

https://github.com/pybamm-team/PyBaMM/blob/develop/pybamm/parameters/parameter_sets.py

tinosulzer, 2022-11-08:
roughly 0.5s in the `__init__` of a total 2s time for `import pybamm`, all in the `.load()` call

awadell1, 2022-11-08:
Oh damn; almost certainly, then. `__init__` can just fetch the list of names and calling `.load()` can be delayed until each parameter set is first accessed in `__getitem__`.

PR #2448 (2022-11-10): Reduce pybamm.parameter_sets init time
- fix: lazily load parameter set entry points
- refactor: Avoid calling dunder methods in parameter_sets

# Description

Lazily load parameter sets instead of doing it on creation, reduces `import pybamm` by ~0.5s

Fixes #2436

## Type of change

Please add a line in the relevant section of [CHANGELOG.md](https://github.com/pybamm-team/PyBaMM/blob/develop/CHANGELOG.md) to document the change (include PR #) - note reverse order of PR #s. If necessary, also add to the list of breaking changes.

- [ ] New feature (non-breaking change which adds functionality [... cut]
closed 2022-11-10 by tinosulzer