pybamm-team/PyBaMM #2439: Autogenerate docstrings

tinosulzer, 2022-11-09:
There's a lot of repetition in the docstrings for expression tree classes and submodel classes, which leads to errors.
Would be cool to automatically generate the docstrings based on the arguments and the parent function, i.e.

- Detect what the arguments of `__init__` are and add the relevant parameter docstring for each argument
- Detect what the parent class is and add `**Extends:**`

Some ideas for how to do this from stackoverflow https://stackoverflow.com/questions/4056983/how-do-i-programmatically-set-the-docstring and https://stackoverflow.com/questions/13603088/python-dynamic-help-and-autocomplete-generation

tinosulzer, 2022-12-19:
Some other potentially useful references for this
- https://stackoverflow.com/questions/67205506/how-to-avoid-repeating-docstring-and-parameters-for-python-subclasses
- https://stackoverflow.com/questions/2025562/inherit-docstrings-in-python-class-inheritance?noredirect=1&lq=1
- https://stackoverflow.com/questions/5516032/docstring-inheritance-for-properties-using-sphinxs-autodoc
- https://stackoverflow.com/questions/8100166/inheriting-methods-docstrings-in-python?noredirect=1&lq=1

agriyakhetarpal, 2022-12-21:
I could give this one a try, though I would need a bit of help

tinosulzer, 2022-12-21:
great, happy to help

agriyakhetarpal, 2022-12-23:
> Detect what the parent class is and add **Extends:**

So I went through all the links that were given, but I think found a simple fix in the [Sphinx AutoAPI documentation](https://sphinx-autoapi.readthedocs.io/en/latest/reference/config.html#confval-autoapi_options), via the `show-inheritance:` directive which automatically picks up the base class for a derived class. By adding it, for example, in the `binary_operator.rst` file 
```
.. autoclass:: pybamm.BinaryOperator
  :members:
  :show-inheritance:
```
and then building the documentation will add the following line just below the name of the associated class like this
```
class pybamm.BinaryOperator(name, left, right)
Bases: [Symbol](http://127.0.0.1:8000/source/expression_tree/symbol.html#pybamm.Symbol)
```

So it writes the word **Bases** rather than **Extends**, (I don't know how to change that though) and links the base class for the derived class as intended. This will have to be manually added for all the autoclass directives similarly wherever it is applicable.
Am I on the right track? If yes, I shall open a PR and proceed to the other part of the issue, i.e.,  figuring out how to determine the arguments of `__init__` and setting them programmatically.

tinosulzer, 2022-12-28:
>  though I would love to know how to approach it before attempting

The links in my second comment should help with this

> Addendum: we can skip adding it manually by adding the following to conf.py, (the Sphinx configuration file)

This looks perfect

closed 2023-03-10 by tinosulzer