pybamm-team/PyBaMM #2456: Use casadi map to more quickly evaluate processed variables

tinosulzer, 2022-11-14:
https://web.casadi.org/docs/#map

jaskiratsingh2000, 2023-02-20:
Hi @tinosulzer ! Does the casadi map has to be used in the place of "base_variable_casadis" because, I feel that that's what getting processed.

tinosulzer, 2023-02-20:
instead of this for loop https://github.com/pybamm-team/PyBaMM/blob/6ab6c15979ad1b11fcebea37e43723179d201ea6/pybamm/solvers/processed_variable.py#L123-L131

jaskiratsingh2000, 2023-02-20:
Hi @tinosulzer Do you thinnk that something this should work for us?

```
for ts, ys, inputs, base_var_casadi in zip(
        self.all_ts, self.all_ys, self.all_inputs_casadi, self.base_variables_casadi
):
    num_times = len(ts)
    num_y_vals = ys.shape[0]
    num_entries = num_times * num_y_vals

    # Create a function to evaluate the base_var_casadi over a list of inputs
    def evaluate_base_var_casadi(t, y, inputs):
        return float(base_var_casadi(t, y, inputs))

    # Create casadi functions for the inputs and the base_var_casadi
    input_func = casadi.Function('inputs', [self.input_time] + self.input_vars, ['input'], ['cdense'])
    base_var_func = casadi.Function('base_var_casadi', [self.input_time, self.input_y] + input_func(self.input_time), ['output'], ['cdense'])

    # Evaluate the inputs
    input_vals = [np.array([t] + list(y)) for t, y in zip(ts, ys.T)]
    inputs_casadi = input_func(casadi.horzcat(*input_vals)).full().T

    # Evaluate the processed variables using casadi.map
    entries = casadi.map(evaluate_base_var_casadi, ts, ys.T, inputs_casadi).full().ravel()

    # Set the entries for this output variable
    self.entries[idx:idx + num_entries] = entries

    idx += num_entries
```

jaskiratsingh2000, 2023-02-21:
@tinosulzer Any updates on above what I mentioned?

jaskiratsingh2000, 2023-02-21:
Hi @tinosulzer Thanks for letting me know. I'm gently trying to understand the deep of PyBaMM that's why I thought to work on such issues along side issues with labelled "easy"

rtimms, 2024-11-22:
closing, see https://github.com/pybamm-team/PyBaMM/pull/4464

closed 2024-11-22 by rtimms