pybamm-team/PyBaMM #2457: Make `jupyter` a dev-only dependency

tinosulzer, 2022-11-15:
There are some issues with dependencies on arm64 (`qtconsole`)

tinosulzer, 2022-11-15:
Further, with the exception of `numpy`, all of our dependencies could be "optional" (i.e. some features don't work if you don't have them installed). For most dependencies this is easy (e.g. `anytree`, `imageio`), for others this will require some refactoring (`casadi`).

The goal eventually is to have a `pybamm-lite` package that only installs numpy and pybamm, and a `pybamm` that installs all the optional dependencies

ayeankit, 2023-02-20:
hey @tinosulzer, I want to work in this issue. Can you please assign me?

deepeshaburse, 2023-03-05:
Hi @tinosulzer, can I work on this issue?

deepeshaburse, 2023-03-05:
@tinosulzer I was looking through some solutions. I was thinking we could make a separate dev-requirements.txt and add `jupyter` to it (and remove it from requirements.txt). Would that be the right approach to solve this issue?

Saransh-cpp, 2023-03-07:
This would require removing `jupyter` completely from `requirements.txt` and moving it from a required dependency to a `dev` dependency in `setup.py`.

From here -
https://github.com/pybamm-team/PyBaMM/blob/f45f4a5988da047ee356a29fc570475b1a110421/setup.py#L191-L209

to here -
https://github.com/pybamm-team/PyBaMM/blob/f45f4a5988da047ee356a29fc570475b1a110421/setup.py#L218-L221

I remember there was some discussion about creating more optional dependency groups (for instance `notebooks` and `plot`, just like `dev`), but I think the message vanished in slack. You might have to wait before there is a consensus on this 🙂

deepeshaburse, 2023-03-07:
> This would require removing `jupyter` completely from `requirements.txt` and moving it from a required dependency to a `dev` dependency in `setup.py`.
> 
> 
> 
> From here -
> 
> https://github.com/pybamm-team/PyBaMM/blob/f45f4a5988da047ee356a29fc570475b1a110421/setup.py#L191-L209
> 
> 
> 
> to here -
> 
> https://github.com/pybamm-team/PyBaMM/blob/f45f4a5988da047ee356a29fc570475b1a110421/setup.py#L218-L221
> 
> 
> 
> I remember there was some discussion about creating more optional dependency groups (for instance `notebooks` and `plot`, just like `dev`), but I think the message vanished in slack. You might have to wait till there is a consensus on this 🙂

Sure, thanks for letting me know ☺️

tinosulzer, 2023-03-07:
Happy to proceed with just moving jupyter to the dev for now. This the largest and most optional dependency so it makes sense to just do that one first

tinosulzer, 2023-03-19:
Here's an example of how pandas does it: https://pandas.pydata.org/docs/dev/whatsnew/v2.0.0.html#installing-optional-dependencies-with-pip-extras

arjxn-py, 2023-03-28:
If no one is working on this issue, may I look into this and try to do it? @tinosulzer @Saransh-cpp

Saransh-cpp, 2023-03-29:
Let's wait for @deepeshaburse's response for a couple of days. If there is no activity then you can take this up.

arjxn-py, 2023-03-31:
@Saransh-cpp I believe she's not responding

arjxn-py, 2023-04-01:
@Saransh-cpp I have opened this PR trying to solve this issue but I am getting some failing tests, I would be glad if any lead could be provided.

PR #2846 (2023-04-03): Issue #2457 make jupyter a dev only dependency
# Description

Made jupyter a development only dependency instead of required dependency.

Fixes #2457 

## Type of change

Jupyter has been moved from required dependency to development only dependency in `requirements.txt` & `setup.py` and also been added as a testing dependency in `tox.ini`.

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Optimization (back-end change that speeds up the code)
- [x] Bug fix (non-breaking change which fixes an issue)

# Key checklist:

- [x] No style issues: `$ pre-commit run` (see [CONTRI [... cut]
closed 2023-04-03 by tinosulzer